openhwfoundation/cva6 #384: Flush signal in PTW

zhaoxiahust, 2020-03-04:
Hi guys, 

  If we have the flush_i signal valid, why cannot we set state_d=IDLE directly? Why do we need the WAIT_RVALID state?
  I read [issue61](https://github.com/pulp-platform/ariane/issues/61) which mentions that without checking we will have an error. However, I still cannot understand how this violates the req, gnt protocol.
  Any comments about this? An example will help a lot!!

Cheers,
Xia

zhaoxiahust, 2020-03-09:
I understand the problem now to some degree. There is an outstanding access from PTW in Dcache. You cannot directly jump to the idle state as the useless access can come back in the future.

zarubaf, 2020-04-15:
Yes exactly!

closed 2020-04-15 by zarubaf