openhwfoundation/cva6 #385: Unprivileged load issue

atishp04, 2020-03-05:
While porting OpenPiton platform to OpenSBI, I observed that continuous unprivileged loads leads to an illegal memory access error. Is it a known issue ?

Steps to reproduce: Run OpenSBI + Linux (with CONFIG_SMP) on OpenPiton platform
https://github.com/atishp04/opensbi/commits/openpiton

The LED6 turns on as soon as /init starts executing which indicates a unmapped memory access error. The above tree also has the hack that explains the issue as well.

https://github.com/atishp04/opensbi/commit/364567e38f600a1ef1f830a3eaab91410afb6db7

FYI: SBI v0.2 doesn't require unprivileged load. So I can boot all the cores in Linux in that case without any issue. However, unprivileged loads should be fixed in Ariane if that's an issue.

atishp04, 2020-03-11:
ping ?

zarubaf, 2020-03-13:
Thanks for bringing that up and no, there is no known issue with unprivileged loads. In fact, we were using it for debugging Linux code. I absolutely agree that this should be fixed.  The first thing would be to create a test case that allows us to trigger the faulty behavior in RTL simulation. I will keep this open as a tracking issue.

atishp04, 2020-03-13:
Great. Thanks. Let me know if you need any help from me.

MikeOpenHWGroup, 2023-02-18:
>  I absolutely agree that this should be fixed.

Hi @zarubaf, what is preventing that from happening?

Jbalkind, 2023-02-18:
When we dug in and debugged this somewhat, I think we found that this was an MMU issue which we then tried to fix with Tianrui's recent PR. I don't think we've continued to see issues of this variety with newer kernels as a result. Not 100% sure though.