openhwfoundation/cva6 #386: Make without FPGA ?

snat99-hub, 2020-03-06:
Hi,

Using the current Makefile as supplied I am getting completion on RISC V ISA Tests like so:

1) $work-ver/Variane_testharness $RISCV_TESTS/isa/rv64um-v-divu
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 58541

/mnt/c/Users/snata/Documents/tools/linuxtools/riscv-tests/isa/rv64ui-p-srlw completed after 101606 cycles
CPU time used: 30359.38 ms
Wall clock time passed: 30408.37 ms

2) However when attempting to run a C test with pk:
 $work-ver/Variane_testharness  pk hello_ariane.elf 

it seems to pretty much hang.

Adding a +jtag_rbb_enable=1  gets a: Attempting to accept client socket   response but still a hang.

spike using the same pk runs fine on hello_world.elf

Q: Is it possible to make and run a Ariane Verilator model without FPGA dependencies ?

snat99-hub, 2020-03-06:
Never mind. Completed after 20 min on a laptop with  i5-6200U CPU (Skylake 14nm, 2 core, 2.3 GHz).

closed 2020-03-06 by snat99-hub

hyf6661669, 2020-04-29:
The reason is that Ariane uses simulated debug host to load all the information from the elf file into the ram. So the loading process is time-consuming and ```hello.elf``` is much larger than ```rv64um-v-divu```.