openhwfoundation/cva6 #387: Performance degradation since 846e297c and bf71095b

ASintzoff, 2020-03-09:
The commits related to the cache and the load unit (846e297c and bf71095b) have big impacts on the performance of Ariane.

Hereafter, some results from riscv-tests/benchmarks starting with commit 55201929 (parent of 846e297c):

Microseconds for one run through Dhrystone: 
- 55201929: 520
- 846e297c: 652
- bf71095b: 927

towers.riscv:
- 55201929: mcycle = 6643
- 846e297c: mcycle = 10952
- bf71095b: mcycle = 12864

vvadd.riscv:
- 55201929: mcycle = 2464
- 846e297c: mcycle = 4855
- bf71095b: mcycle = 5154

qsort.riscv:
- 55201929: mcycle = 238141
- 846e297c: mcycle = 241890
- bf71095b: mcycle = 367753

rsort.riscv:
- 55201929: mcycle = 224689
- 846e297c: mcycle = 268456
- bf71095b: mcycle = 409953

zarubaf, 2020-03-13:
Thanks for the detailed report. It can certainly be that there is a performance regression introduced, but right now I am a bit inclined to say that this is due to the fact that all those programs are run in debug mode (which exhibits uncacheable and non-idempotent loads which are taking longer after this commit). 
I couldn't see a notable regression during Linux boot but I might need to double-check that. It might also be a good idea to look more detailed at the generated traces. Thanks for bringing this up, I will keep it open for further investigations.

zarubaf, 2023-10-20:
Closing because of stale-ness. We track performance in the CI these days, any findings from back in the day are highly unlikely to lead us to useful information today.

closed 2023-10-20 by zarubaf