openhwfoundation/cva6 #390: The static library: libfesvr.so & libfesvr.a

Yi1101, 2020-03-11:
when I try to run the following command, the error shows can't not find this file $(RISCV)/lib/libfesvr.so.

>make sim elf-bin=$RISCV/riscv64-unknown-elf/bin/pk target-options=hello.elf 

Then I try to change Makefile "$(RISCV)/lib/libfesvr.so" to "$(RISCV)/lib/libfesvr.a", then the error disappears, but another errors show as follows.

 Time: 0 ns  Iteration: 0  Unknown: <UNKNOWN>
# Loading /home/chen/code/riscv-gnu-toolchain/lib/libfesvr.a
# ** Error** (suppressible): (vsim-3197) Load of "/home/code/riscv-gnu-toolchain/lib/libfesvr.a" failed: /home/chen/code/riscv-gnu-toolchain/lib/libfesvr.a: invalid ELF header.
# ** Warning: (vsim-3116) Problem reading symbols from /home/code/riscv-gnu-toolchain/lib/libfesvr.a : Not an ELF file.
# ** Warning: ** Warning: (vsim-3772) Failed to find the undefined symbols for shared library file "/home/chen/code/riscv-gnu-toolchain/lib/libfesvr.a".

Many thanks in advance!

Moschn, 2020-03-11:
I believe we still rely on an old version of riscv-fesvr. Nowadays, `riscv-fesvr` is packaged with spike in `riscv-isa-sim`. However, we did not update our paths yet. Try to install an old version of `riscv-fesvr` like the one from ariane-sdk.

Yi1101, 2020-03-11:
Thank you for your prompt reply. However, I still have one question. I use the followed two commends to convert the file  "libfesvr.a" into file "libfesvr.so", and the error disappear. So the question is, using this method to get the file "libfesvr.so" has any difference with the method you have recommended. 
<ar -x libfesvr.a
<gcc -shared *.o -o libfesvr.so

Moschn, 2020-03-11:
I do not know. If everything works then I guess it is fine.

closed 2020-03-11 by Moschn