openhwfoundation/cva6 #405: failed to run Dhrystone on Verilator

AnouarNechi, 2020-04-03:
I want to run Dhrystone in the verilator I compiled the code from riscv-tests as following:
`$ riscv64-unknown-elf-gcc -I./../env -I./common -I./dhrystone -DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O2 -ffast-math -fno-common -fno-builtin-printf -o dhrystone.elf ./dhrystone/dhrystone.c ./dhrystone/dhrystone_main.c ./common/syscalls.c ./common/crt.S -static -nostdlib -nostartfiles -lm -lgcc -T ./common/test.ld`
Than I run the verilator:
```
$ make verilate
$ work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf
```
I got the following error : 
```
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 36781
bbl loader
couldn't open ELF program: dhrystone.elf!
/home/nechi/Documents/ariane-sdk/install/riscv64-unknown-linux-gnu/bin/pk *** FAILED *** (code = -1) after 2300345 cycles
CPU time used: 568687.99 ms
Wall clock time passed: 568793.25 ms
```

Yi1101, 2020-04-06:
I have the same problem as you. 
>./work-ver/Variane_testharness riscv64-unknown-elf/bin/pk dhrystone.riscv
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 45823
bbl loader
couldn't open ELF program: dhrystone.riscv!
/home/chen/code/riscv-gnu-toolchain/riscv64-unknown-elf/bin/pk *** FAILED *** (code = -1) after 2128253 cycles
CPU time used: 467425.49 ms
Wall clock time passed: 467592.57 ms

zarubaf, 2020-04-14:
Does it work in `spike`? Seems to me like a SW issue...
```
spike work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk dhrystone.elf
```

axeldbom, 2020-05-15:
I have the same problem and when I run 
`spike work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk dhrystone.riscv`

I get the error:
`spike work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk dhrystone.riscv
 ERROR: invalid load from debug module: 8 bytes at 0x00000000003fa380
 terminate called after throwing an instance of 'trap_load_access_fault'`

zarubaf, 2020-05-15:
This indicates that something is wrong with the binary. If this is the `dhrystone.riscv` from the `riscv-tests` repository I thought to remember that they are bare-metal only. You would need to re-compile dhrystone using the standard linker settings for user-space applications.

Yi1101, 2021-02-03:
How did you slove this problem finally? I tried to add different compiled options but the results were the same. 
When run
```
make sim elf-bin=$RISCV/riscv64-unknown-elf/bin/pk target-options=dhrystone.riscv

```
But the phenomenon showed as followed, in the questasim transcript, it showed:
```
# couldn't open ELF program: dhrystone.riscv!
# UVM_ERROR tb/ariane_tb.sv(111) @ 55384050: reporter [Core Test] *** FAILED *** (tohost = 2147483647)
```
Why run ` make sim `has this error, but the dhrystone.riscv worked fine with `work-ver/Variane_testharness dhrystone.riscv`?

sebastien-riou, 2021-08-17:
if you follow the instructions in riscv-tests, this works fine:
````
spike $RISCV/target/share/riscv-tests/benchmarks/dhrystone.riscv
Microseconds for one run through Dhrystone: 392
    Dhrystones per Second:                      2550
    mcycle = 196024
    minstret = 196029
````
and this as well:
````
work-ver/Variane_testharness $RISCV/target/share/riscv-tests/benchmarks/dhrystone.riscv
    This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
    Listening on port 37763
    Microseconds for one run through Dhrystone: 522
    Dhrystones per Second:                      1915
    mcycle = 261116
    minstret = 196029
    /opt/riscv/target/share/riscv-tests/benchmarks/dhrystone.riscv completed after 435434 cycles
    CPU time used: 48697.94 ms
    Wall clock time passed: 49024.01 ms
````

dont't know about QUESTASIM but hey, this issue mention specifically verilator in the title, consider opening a new one...

closed 2023-02-18 by MikeOpenHWGroup