openhwfoundation/cva6 #406: could not initializing SD

shirlynan, 2020-04-05:
Hello,
I am trying to boot linux on Ariane SoC on VCU118.
I've followed the 'Booting from SD card' steps from ariane-sdk and used the bbl.bin file not only self-generated from sdk but also the one stored under https://github.com/pulp-platform/ariane-sdk/releases . pull-up risistor adater is also used for PMOD interface.
However I got the terminal outputs:
Hello World!
init SPI
status: 0x0000000000000025
status: 0x0000000000000025
SPI initialized!
could not initialize sd... exiting
What does this actually mean? Where did I wrong?

zarubaf, 2020-04-14:
I think the SPI master has problems communicating with the SD card. Seems to be a board problem (maybe timing violations on the VCU118). I remember that the PMOD SD card module was very brittle timing-wise.

MikeOpenHWGroup, 2023-02-18:
Thanks for your interest in CVA6 @shirlynan!

This issue has not been updated for almost three years, so I will close it.  If you are still having problems, either re-open it (with additional information) or create a new issue.

closed 2023-02-18 by MikeOpenHWGroup