openhwfoundation/cva6 #408: The exception and D-cache return data in Load unit

zhaoxiahust, 2020-04-08:
Hi guys,
  In the load unit, when the exception and D-cache return data come together, what will happen? It seems to me that the exception and data will be both written back to the slot of the mem_q in the scoreboard unit. This slot actually belongs to the D-cache return data. Will this cause a problem when this slot of the mem_q is committed? Did I miss anything?

  Any help is welcome!

Cheers,
Xia

zarubaf, 2020-04-15:
If the exception valid flag is set in the scoreboard entry `mem_q`, the commit stage will not commit the data.

closed 2020-04-15 by zarubaf

zhaoxiahust, 2020-04-15:
Yes，I understand this will not cause any errors as the instruction which is falsely marked as an exception will be executed again. But is it a bug as the instruction should not be marked as exception fault.