openhwfoundation/cva6 #409: fcvt.s.d gives wrong answer

b1f6c1c4, 2020-04-10:
I tried to run `fcvt.s.d` to convert a small double (`2.386004908190509275e-40` which is `0x37b4c8f800000000`) into single, but it gives `inf` (`0x7f800000`) instead of `2.38600491e-40` (`0x0002991f`).

stmach, 2020-04-10:
Can you please provide a minimal working code snippet that you execute on Ariane which triggers this issue, so that we may reproduce it?

b1f6c1c4, 2020-04-10:
I saw this error when I was trying to run riscv-tests (https://github.com/riscv/riscv-tests/blob/master/isa/rv64ud/recoding.S) with OpenPiton.

zarubaf, 2020-04-14:
Can you check the signals of the FPU, please? 
@b1f6c1c4 and @stmach can we check that you tested the stimuli on the same fpnew source?

stmach, 2020-04-14:
I've used the latest tagged release `v0.6.1`, of the FPU but there haven't been changes to the conversion units in a long time, so it should work with older versions as well.

zarubaf, 2020-04-14:
We have recoding in the CI, so I don't think it fails for the repo here anyway.

b1f6c1c4, 2020-04-14:
Here's what I got. BTW, I'm using fpnew `v0.6.1` (`4a241d4`).

[image]

[image]

zarubaf, 2020-04-15:
@stmach Can this be a VCS/Modelsim difference? When compiling the FPU I am getting two warnings with the FPU in VCS so that might be related (but I don't think so):
```
Warning-[ENUMASSIGN] Illegal assignment to enum variable
/scratch/zarubaf/snitch/src/fpnew/src/fpnew_opgroup_block.sv, 126
fpnew_opgroup_block, "fpnew_pkg::get_first_enabled_multi(10'b1010101010, FpFmtMask)"
  Only expressions of the enum type can be assigned to an enum variable.
  The type bit [9:0] is incompatible with the enum 'unit_type_t'
  Expression: 10'b1010101010
  Use the static cast operator to convert the expression to enum type.

Warning-[ENUMASSIGN] Illegal assignment to enum variable
/scratch/zarubaf/snitch/src/fpnew/src/fpnew_opgroup_block.sv, 126
fpnew_opgroup_block, "fpnew_pkg::get_first_enabled_multi(10'b1010101010, FpFmtMask)"
  Only expressions of the enum type can be assigned to an enum variable.
  The type bit [9:0] is incompatible with the enum 'unit_type_t'
  Expression: 10'b1010101010
  Use the static cast operator to convert the expression to enum type.
```
`Bender` supports generating VCS scripts now and its relatively easy to bring up the simulation in VCS as well.

zarubaf, 2020-04-15:
@b1f6c1c4 What's your VCS version? Just to be sure we are not hunting bugs in the FPU although they are in the CPU (wrapper for example). Can you provide us with the same screenshot but signals from the `fpnew_top` instance (https://github.com/pulp-platform/ariane/blob/9cebe8a238aa8e1450c085f051fefc1e6d55cc41/src/fpu_wrap.sv#L513-L538)?

stmach, 2020-04-15:
I tried to run the operation in VCS and indeed i receive Inf (`0x7f800000`).
@zarubaf I added a static cast to `int` to solve your warning so it shouldn't be the reason for this mismatch. Feel free to move this issue back to `fpnew`.
I'll try to find out what triggers VCS to produce the wrong result here... It might take a while though as VCS is not our main simulation target.

b1f6c1c4, 2020-04-15:
FYI
```
vcs script version : O-2018.09
machine name = della2.princeton.edu
machine type = linux64
machine os = Linux 3.10.0-1062.12.1.el7.x86_64
The FLEXlm host ID of this machine is "80000208fe80 00219ba4ae77 00219ba4ae79 00219ba4ae7b 00219ba4ae7d"
Compiler version = VCS-MX O-2018.09-SP2_Full64
VCS Build Date = Feb 28 2019 22:34:30
```