openhwfoundation/cva6 #410: Is it safe to disable and enable data cache at runtime

jimmysitu, 2020-04-11:
Hi, All

I am trying to work Ariane SoC with a SD controller, which has a DMA engine in it. The cache I use is write-through cache. I need to flush the D cache before DMA move SD blocks to DRAM.
As I found in #355 #339, the fence instruction do not really flush the write-through cache but only the write buffer. I think a better way is to disable the whole cache and re-enable it.
I try it on my FPGA with
```c
    __asm__ volatile("csrwi 0x701, 0x00");
    __asm__ volatile("csrwi 0x701, 0x01");
```
It seems work, but I am not sure if there is any potential risk. Any comment?

Thanks

zhaoxiahust, 2020-04-13:
If you disable the cache and write a data to memory. Then, you enable the cache and try to read the data again. If the old data exists in the cache, it will cause an error. This is the error from the arch view. I am not sure the specific implementation in Ariane will cause other errors or not.

zarubaf, 2020-04-14:
I think this works for now as the cache when enabled is flushing the cache. This is, unfortunately, an artifact of the missing coherence support in the single-core Ariane version.

MikeOpenHWGroup, 2023-02-18:
Hi @jimmysitu, thanks for your interest in CVA6.  As @zarubaf, points out this, this works for now, and there have been no comments for almost three years, so I will close this issue.  If you disagree, please re-open the issue.   Thanks.

closed 2023-02-18 by MikeOpenHWGroup