openhwfoundation/cva6 #422: Connect new slave devices with crossbar

MasterMofa, 2020-04-30:
I tried to insert a new slave device (such as UART module) into the ariane soc by connecting them on cross bar ,however, the AXI crossbar becomes disabled after I modified the number of peripherals(NB_peripherals) and address range. I am wondering if I can mount a new device on crossbar of ariane.
  Additionally, I am tring to replace the origenial crossbar by Xilinx crossbar,but I found that an axi signal named "aw_atop" ,which is not included in the standard axi protocol. Can ariane cpu still work if I replace its crossbar by a Xilinx standard cross bar/interconnect core? What is the functionality of this aw_atop. Would you please foward me some instruction on it. I would appreciate it.

zarubaf, 2020-05-15:
In addition to attaching the new slave device, you will need to make sure to provide an address entry for it, and increase the number of master ports on the crossbar (everything in the `ariane_soc_pkg.sv`).

Yes, `aw_top` is defined in AXI5. This adds the capabilities to resolve atomic memory operations close to the memory and it is needed for Ariane (except if you don't need any atomic memory operations).

MasterMofa, 2020-05-22:
Thank Q for your preious reply, the corssbar is already replaced, after I replaced originial crossbar by xilinx axi interconnect，ariane can still run 0 stage bootloader ，but it cannot boot system though the data in the Memory（DDR，address 0x80000000）is correct. 
I tried to fliter the atomic signal by this one: [https://github.com/pulp-platform/axi/blob/master/src/axi_atop_filter.sv](url),but nothing chenges.
Wether the aw_atop function of ariane core can be disabled by modify some sv files?  Alternatively, is there any manner to turn of the atomic memory operations by arrange the vmlinux and bbl settings?To put it another way , what should I do to make whole ariane work correctly without atomic memory operations?

farnam16, 2021-03-17:
Hello, I am trying the same. I attached an AXI Lite Slave port to the AXI Crossbar. I updated the ariane_soc_pkg.sv as follows: 

  typedef enum int unsigned {
    DRAM     = 0,
    GPIO     = 1,
    Ethernet = 2,
    SPI      = 3,
    UART     = 4,
    PLIC     = 5,
    CLINT    = 6,
    ROM      = 7,
    Debug    = 8,
    Xsm      = 9  // farnam: xsm master peripheral port on the AXI crossbar, to attach to the xsm registers. 
  } axi_slaves_t;

 
  localparam NB_PERIPHERALS = Xsm + 1; 
  localparam logic[63:0] XsmLength      = 64'h10000; 
  
    typedef enum logic [63:0] {
    DebugBase    = 64'h0000_0000,
    ROMBase      = 64'h0001_0000,
    CLINTBase    = 64'h0200_0000,
    PLICBase     = 64'h0C00_0000,
    UARTBase     = 64'h1000_0000,
    SPIBase      = 64'h2000_0000,
    EthernetBase = 64'h3000_0000,
    GPIOBase     = 64'h4000_0000,
    XsmBase      = 64'h5000_0000,
    DRAMBase     = 64'h8000_0000
    // farnam: xsm registers base address. 
  } soc_bus_start_t;
  
  Unfortunately, when I update the axi_bar start_addr and end_addr inputs respecting new memory map set in ariane_soc_pkg.sv  the riscv hangs and UART doesn't print in the terminal screen. Do I miss some other parameters to be set? I have to add that I have not yet connect the AXI Lite Slave port into the Axi Crossbar, and freezing happens just when I add the new addresses into the axi cross bar instances In ariane_xilinx.sv.  Thank you for your helps in advance.

farnam16, 2021-06-23:
@farnam16 
Dear all, 

I solved my issue: 

It is important that keep the order of the peripheral as it is instantiated in the address range of the AXI Crossbar. 
I wrongly assigned the enum type definition for my custom module (here named Xsm). I fixed it by putting it between the 
DRAM = 0 and GPIO = 2 as Xsm = 1. I reordered the type enum like here: 

  typedef enum int unsigned {
    DRAM     = 0,
    BRAM     = 1,  // farnam: BRAM master peripheral port on the AXI crossbar, to attach to the BRAM
    GPIO     = 2,
    Ethernet = 3,
    SPI      = 4,
    UART     = 5,
    PLIC     = 6,
    CLINT    = 7,
    ROM      = 8,
    Debug    = 9
  } axi_slaves_t;
   
localparam NB_PERIPHERALS = Debug + 1;

Note: in the AXI crossbar module instantiation in the ariane_xilinx.sv also the order should be respected: i.e,: 

    .start_addr_i ({
        ariane_soc::DebugBase,
        ariane_soc::ROMBase,
        ariane_soc::CLINTBase,
        ariane_soc::PLICBase,
        ariane_soc::UARTBase,
        ariane_soc::SPIBase,
        ariane_soc::EthernetBase,
        ariane_soc::GPIOBase,
        ariane_soc::XsmBase,
        ariane_soc::DRAMBase

    }),
    .end_addr_i   ({
        ariane_soc::DebugBase    + ariane_soc::DebugLength - 1,
        ariane_soc::ROMBase      + ariane_soc::ROMLength - 1,
        ariane_soc::CLINTBase    + ariane_soc::CLINTLength - 1,
        ariane_soc::PLICBase     + ariane_soc::PLICLength - 1,
        ariane_soc::UARTBase     + ariane_soc::UARTLength - 1,
        ariane_soc::SPIBase      + ariane_soc::SPILength - 1,
        ariane_soc::EthernetBase + ariane_soc::EthernetLength -1,
        ariane_soc::GPIOBase     + ariane_soc::GPIOLength - 1,
        ariane_soc::XsmBase      + ariane_soc::XsmLength - 1,
        ariane_soc::DRAMBase     + ariane_soc::DRAMLength - 1
        
    }),

Saluti,
fkhm

Bill94l, 2021-06-25:
Hi @farnam16 
how did you connect the axi-lite with the corsbar (Xbar) ?
Thnaks

farnam16, 2021-06-25:
@Bill94l 
It is very simple. You simply define a slave port in your wrapper code by: 
AXI_BUS.Slave                       wrapper_axi_slave,
Then, you just need to connect your AXI slave signals to their corresponding signals packed into the  "your_axi_slave"

For instance: 

assign your_axi_slave_awaddr =  wrapper_axi_slave.aw_addr;//IN
assign your_axi_slave_awvalid = wrapper_axi_slave.aw_valid;//IN

assign wrapper_axi_slave.aw_ready = your_axi_slave_awready;//OUT

 
 Finally, in the top module (ariane_xilinx.sv) from your wrapper instantiation you connect the wrapper axi slave port to one of the AXI Cross bar master ports: 
 
 For example: 
 
 .wrapper_axi_slave  ( master[ariane_soc::your_defined_module_enum_name_in_ariane_soc_pkg]), 
  
  Hope this helps
  Saluti
  f

farnam16, 2021-06-25:
hi @Bill94l 

Note that you should also update the ariane.dts with the information of your new device, and re-make (make all in the folder of /fpga/bootrom)  the bootrom for new bootrom.sv file which is necessary for your devices to be appeared into the OS kernel. Then you can have a kernel module to map its physical address to the virtual address, through which in the program you access to the device.

Bill94l, 2021-06-25:
Thank you so much @farnam16

farnam16, 2021-06-25:
> Thank you so much @farnam16

your welcome.

Bill94l, 2021-08-09:
Dear @farnam16 
Could you take a look at my issue [https://github.com/openhwgroup/cva6/issues/703](url) which is to add a new IP on the bus, maybe you will have a suggestion?
Thanks in advance

monireee, 2022-03-05:
> hi @Bill94l
>
> Note that you should also update the ariane.dts with the information of your new device, and re-make (make all in the folder of /fpga/bootrom) the bootrom for new bootrom.sv file which [...]

I noticed that ariane.dts for Linux is broken. Would you please share it in this thread if you have? Thank you.

farnamatic, 2022-03-05:
Hi, 

Please use the official one in the [https://github.com/openhwgroup/cva6/tree/master/corev_apu/fpga/src/bootrom](url). It is under the cv64a6.dts or cv32a6.dts names.

MikeOpenHWGroup, 2023-02-18:
@MasterMofa, @Bill94l, @farnamatic and @monireee, thanks for your interest in CVA6.  This issue is more than 2.5 years old, has no pull-requests linked to it and has not received any updates for almost one year.  Furthermore it seems to be a discussion about adding new functions to locally modified versions of `corev_apu`.  Given that, I will close this issue now.   If the problem still persists, please open a new issue.  Thanks!

closed 2023-02-18 by MikeOpenHWGroup