openhwfoundation/cva6 #424: Enable to run User-Space applications

SpaceNitro, 2020-05-07:
Hello,

When I follow the steps to run "User-Space applications" from Ariane directory, I get the error : ../configure : no file or directory 
when I try to use this command : ../configure --prefix=$RISCV --host=riscv64-unknown-elf
[image]

Riscv-gnu toolchain, riscv-tools, verilator are installed.

Could you help me to solve this error ? 
Thank you for your time and consideration.

zarubaf, 2020-05-15:
That step should be done in `riscv-pk` not in the build folder of Ariane.

zarubaf, 2020-05-15:
Once you have `pk` installed in your RISC-V install dir you can use it to run user-space applications.

SpaceNitro, 2020-05-29:
Hello, 
Thank you for your answer.

I installed riscv-pk in the install directory. But when I try to run the make file, I get an error :
[image]
[image]

What is wrong with my installation?
Thank you.

zarubaf, 2020-05-29:
I was running into a similar issue on some machines. The problem is that the `Makefile` tries to use your system (x86) `gcc` to compile a RISC-V binary. You can solve the issue by specifying which compiler to use:
```
make CC=riscv64-unknown-elf-gcc
```

That should solve the issue.

SpaceNitro, 2020-05-29:
I tried `make CC=riscv64-unknown-elf-gcc` but it is not working. 
[image]
[image]

I think the compiler doesn't know where to find `riscv64-unknown-elf-gcc`.
checking for riscv64-unknown-elf-g++... no
checking for riscv64-unknown-elf-c++... no
checking for riscv64-unknown-elf-gpp... no
checking for riscv64-unknown-elf-aCC... no
checking for riscv64-unknown-elf-CC... no
checking for riscv64-unknown-elf-cxx... no
checking for riscv64-unknown-elf-cc++... no
checking for riscv64-unknown-elf-cl.exe... no
checking for riscv64-unknown-elf-FCC... no
checking for riscv64-unknown-elf-KCC... no
checking for riscv64-unknown-elf-RCC... no
checking for riscv64-unknown-elf-xlC_r... no
checking for riscv64-unknown-elf-xlC... no

What is wrong with my riscv64 gcc elf?

SpaceNitro, 2020-06-15:
I managed to solve my problem. My `PATH` variable was not redirected to the bin directory where we can find riscv-unknow-elf-gcc file for example. Thank you for the help.

closed 2020-06-15 by SpaceNitro