openhwfoundation/cva6 #425: icache unknown when ISA extension "C" is enabled

zetalog, 2020-05-11:
Suffering from lots of
{code}
Offending '((|icache_dreq_o_data) !== 1'hx)'
{code}
Will this lead to some bad side effects?

And we can notice that when ISA extension "C" is enabled:
{code}
assign btb_prediction_shifted[0] = (serving_unaligned) ? btb_q : btb_prediction[0]
{code}
And we can see btb_q reset to 0. This looks wrong to me. Say, if we have code block booting from 0x8000_0000, why a default branch prediction result is not the vector entry or a reset vector?

zarubaf, 2020-05-15:
The predictor needs to be trained at a certain point. Ariane has a `PMA` configuration where you restrict the addresses to valid ranges of your SoC. Furthermore, if the predictor is untrained and does not contain any valid data it will not use the prediction mechanism. 

How/where did you enable those checks?

zetalog, 2020-07-26:
We found a problem in Verilog hex conversion, which swallows last bytes of then .data section.
After fixing this issue, so far, I didn't see the warnings popped up.

So let me close it.
Thanks for the explanation.

zetalog, 2020-08-04:
It looks the issue still persists.
When multiple function calls are invoked, descending beyond the RAS depth.
Final return could always cause a default prediction result, using address 0.
And 0 is indeed a valid memory address in SoC. For example:
0 belongs to an address range of DDR, however the CPU is currently running a driver to bring up the DDRs.

zetalog, 2020-08-04:
I also tried to remove DDR region from ExecuteRegion in include/ariane_pkg.sv.
The problem can still reproduce.

MikeOpenHWGroup, 2023-02-18:
Thanks for your interest in CVA6 @zetalog.

Hi @zarubaf. Can you move this forward?

zarubaf, 2023-02-19:
I am not sure what the „bug“ is exactly. In any case, the predictor trains itself and there are no runtime PMAs in CVA6.

zarubaf, 2023-10-20:
Closing because of staleness.

closed 2023-10-20 by zarubaf