openhwfoundation/cva6 #426: One question about the cache_ctrl.sv

zhaoxiahust, 2020-05-14:
Hi guys,
  In the cache_ctrl.sv, when the state is in the "WAIT_CRITICAL_WORD" and there is a new request, Ariane will run the below code. 

                         // Wait until we have access on the memory array
                        if (gnt_i) begin
                            state_d = WAIT_TAG;
                            mem_req_d.bypass = 1'b0;
                            req_port_o.data_gnt = 1'b1;
                        end
   I think to be precise, the code might need to be updated like below?

                         if (gnt_i) begin
                            state_d = WAIT_TAG;
                            mem_req_d.bypass = 1'b0;
                            // only for a read
                            if (!req_port_i.data_we)
                                req_port_o.data_gnt = 1'b1;
                        end

  I understand the current version will not cause any errors as the memory controller is binded to the load or store unit. In fact, only the loads can reach the "WAIT_CRITICAL_WORD" state. Any comments?

Many thanks
Xia

zarubaf, 2020-05-15:
Yes, exactly. The critical word feature is only valid for loads as where the missing word in the cache-line is retrieved first.

MikeOpenHWGroup, 2023-02-18:
Thanks for your interest in CVA6 @zhaoxiahust (and good catch!).

Hi @zarubaf.   You've indicated agreement with Zhaoxiahust's assessment, but as of `a6075fd` the code in question has not been updated.   Can you move this forward?

closed 2023-02-19 by zarubaf