openhwfoundation/cva6 #428: Black box —Design 'unread'

Yi1101, 2020-05-18:
Dear Dr. Zaruba,
When run the synthsis of Ariane core with Design_vision, the check_design command shows the following warning.
> Designs                                                             1

> Black box (LINT-55)                                             1

> Information: Design 'unread' does not contain any cells or nets. (LINT-55)

According the RTL of module unread, which as following.

// Description: Dummy circuit to mitigate Open Pin warnings
/* verilator lint_off UNUSED */
module unread (
    input logic d_i
);
endmodule
/* verilator lint_on UNUSED */

In order to avoid this Black box warning, so in my opinion, the Module unread should be deleted when do the silicon implementation.  Am I right ?

Yi1101, 2020-05-18:
What' more, what is the function of this unread module ? Looking forward to your reply !

zarubaf, 2020-06-15:
It just to tie-off signals which are not read (which gives another warning). For implementation you can ignore the lint warning, or better you try to disable it only for the `unread` module.

closed 2020-06-15 by zarubaf