openhwfoundation/cva6 #429: ICache Refill in Case of an Exception

mofadiheh, 2020-05-19:
Hello, 

It seems here in the ICache that, if a request causes a TLB hit and a Cache miss, the ICache does not check the fetch exception before making a cache refill when it is in the TAG_CMP state:
https://github.com/pulp-platform/ariane/blob/733a283798b07e7f97b483f1856b34d594e0f928/src/cache_subsystem/std_icache.sv#L271-L298

The fetch exception is only checked when the controller is in WAIT_ADDRESS_TRANSLATION before going to the REDO_REQ state. However, if the address is a TLB hit, and therefore the ICache controller does not go to the WAIT_ADDRESS_TRANSLATION, then the exception will be only forwarded to the core. This is generally not a problem unless it is a cache miss, which causes a cache line refill for an address which was not accessible to the currently running process. It may not be considered a bug but it may have security implications regarding side channel attacks. 

Is it the intended behavior of ICache? Am I missing something about this scenario?

Thanks in advance.
Mo Fadiheh

zarubaf, 2020-07-01:
Thanks for pointing that out! Indeed, (while I am absolutely no security expert) there are several side- and [other channels](https://arxiv.org/abs/2005.02193) in Ariane. The design focus wasn't on security in the first place.

MikeOpenHWGroup, 2023-02-18:
Hi @zarubaf, the ball is in your court on this issue.  Is this security issue real?  Are we going to fix it?

zarubaf, 2023-02-19:
That is not my call. I personally will not fix it. But others might want to. So I would leave this open.

MikeOpenHWGroup, 2023-02-27:
Hi @zarubaf, if you remove yourself as Assignee (which is a legitimate thing to do), please assign the issue to someone else.  An unassigned issue is a orphan that will not move forward.

Hi @JeanRochCoulon and @jquevremont, I am giving you this hot-potato.

JeanRochCoulon, 2023-03-16:
@ASintzoff The exception is not generated in that tricky use case. This should be included to the TRAP verification. That's why I forward you the hot-potato.