openhwfoundation/cva6 #430: Cache Controller

mofadiheh, 2020-05-19:
Hello, 

Consider the following scenario: the load_unit is in the IDLE state and req_port_i.data_gnt is zero. A request comes and as a result, the load unit goes to the WAIT_GNT state: 
https://github.com/pulp-platform/ariane/blob/733a283798b07e7f97b483f1856b34d594e0f928/src/load_unit.sv#L102-L115

In the next clock cycle, req_port_o.data_req is high. Let's assume in this clock cycle the cache is available again, so the req_port_i.data_gnt goes high. At the same time, since the cache is available, the corresponding cache controller will go to the WAIT_TAG state in the clock cycle after that: 
https://github.com/pulp-platform/ariane/blob/733a283798b07e7f97b483f1856b34d594e0f928/src/cache_subsystem/cache_ctrl.sv#L129-L163
If at the same clock cycle, the MMU returns an exception to the load unit for the requested address, the load unit will drop the request and goes back to IDLE, while its corresponding cache controller will be in WAIT_TAG, waiting for the physical address tag:
https://github.com/pulp-platform/ariane/blob/733a283798b07e7f97b483f1856b34d594e0f928/src/load_unit.sv#L241-L243

To make the long story short: what would happen if when the load unit is waiting for req_port_i.data_gnt, both data_gnt and exception arrive at the same time? Does it cause the cache controller to proceed with a canceled transaction? 
If so, then it can cause performance degradation and/or security implications.

Many thanks in advance.
Mo Fadiheh

hyf6661669, 2020-05-25:
I agree with you, if ```ex_i.valid & valid_i == 1```, then ```req_port_o.kill_req``` should be set to 1 to prevent the dcache being affected by the load instruction with exception.

zhaoxiahust, 2020-06-01:
If the exception comes, the instruction will be marked with the exception. When the exception is processed in the commit stage, the pipeline will be flushed and the pc will be reset to the instruction which caused the exception. Thus, even though the dcache is affected by the load instruction with exception and the later instructions go wrong because of this, it will not cause any errors as all these will be flushed. 

BTW, I agree this might be a security problem as the exception can be caused by the memory access permission violation. In this case, the cache line might be fetched back to the cache and leaks secure information under the side-channel attack.

MikeOpenHWGroup, 2023-02-18:
Hi @JeanRochCoulon and @zarubaf.  This issue _sounds_ serious, but I am not in a position to make a determination.  Having said that, there has been no comments on it for more than 2.5 years.  Can one of you move this forward?

zarubaf, 2023-10-20:
Should be tracked in #691, closing because of duplication.

closed 2023-10-20 by zarubaf