openhwfoundation/cva6 #434: no default value for ip_re_o in plic_regmap.sv

RaphaelKlink, 2020-05-31:
There is no default value for the ip_re_o signal in plic_regmap.sv.

it is only set for the correct address but there is no value for other address.
Can it be hardwired to 1 or should it be 0 for every other address.

zarubaf, 2020-05-31:
I am not sure I can follow. There is a default for all cases, isn't it?

https://github.com/pulp-platform/rv_plic/blob/ebe3e9888920f1f757b75bab81dc80f804896f57/rtl/plic_regmap.sv#L31

RaphaelKlink, 2020-05-31:
it is not the prio_re_o but the ip_re_o signal. For this one the default value is missing

https://github.com/pulp-platform/rv_plic/blob/ebe3e9888920f1f757b75bab81dc80f804896f57/rtl/plic_regmap.sv#L8

zarubaf, 2020-05-31:
Ah sorry, good catch. I think it doesn't matter actually what the value is (but not inferring a latch would be a good start ;-)) https://github.com/pulp-platform/rv_plic/blob/ebe3e9888920f1f757b75bab81dc80f804896f57/rtl/plic_top.sv#L93

I think a good next step would be to try to get rid of our PULP plic repo altogether and only use the lowrisc one with the proper AXI adapter inferred.

zarubaf, 2020-05-31:
That's a bit of legacy when the register generation infrastructure wasn't open source yet.

RaphaelKlink, 2020-06-01:
Without vivado i would not have it found, was checking the messages for my own code and that one also warrented a warning.

zarubaf, 2020-06-01:
Would you mind just fixing the warning in the plic repo with a pull request? I think just tying it to 0 wl be enough. That would be very appreciated.

MikeOpenHWGroup, 2023-02-18:
Thanks for this issue @RaphaelKlink.

Hi @zarubaf, based on your comments this is a real issue with a known fix.  Was the fix implemented?

zarubaf, 2023-10-20:
This has either been fixed or isn't relevant because we focus on the core here and do not support peripherals. There is a dedicated [`plic` repository](https://github.com/lowRISC/rv_plic). In case the issue persists, please re-open there.

closed 2023-10-20 by zarubaf