openhwfoundation/cva6 #435: Segmentation fault(core dumped) during make verilate

NazerkeT, 2020-05-31:
Dear Ariane Team, I have installed `riscv-gnu-toolchain`, `riscv-tool` with the latest versions available, updated `Ariane` submodules recursively, installed `Verilator` with the latest (4.x) version globally, unset `$VERILATOR_ROOT` by every built, have passed Verilator examples successfully and so far it seems that I have done everything right following manuals. However, for my Ubuntu 18.04 OS, I can't `make verilate` with `Segmentation fault (core dumped) %Error: Command Failed` error. There are plenty of warnings on pin assignments, delays and expected values which you may see in the attached terminal log file. 

Could you help me with the solution of this problem? Initially, I thought that it could be because of my RAM size (6GB) being small, however, it seems it is not the case. 
I have googled around old issues, but for now, I did not find anything useful.
Hope to hear back from you soon.

[make_verilate_log.txt](https://github.com/pulp-platform/ariane/files/4707993/make_verilate_log.txt)

zarubaf, 2020-05-31:
I think that issue is more related to Verilator than to specifically Ariane, you might want to ask at the Verilator Github issue tracker. But I'd try to fire up gdb and see where the segfault happens and try to produce a minimal example. I thought to remember that there was one particular Verilator version that broke with Ariane, so just installing the latest version could already be helpful.

NazerkeT, 2020-05-31:
Thanks for the quick reply! I will approach the Verilator Github page and also be glad to hear back on gdb results. Regarding the Verilator version on my machine, it is `Verilator 4.035 devel rev v4.034-173-gb60a92ee` and seems to be the latest already. If it is the case, do you remember, whether was that broken version similar to something 4.x or different? 

The thing is that, because verilator template examples run successfully already and I was not able to test Ariane yet due to this error, I decided to approach Ariane first. Hope to solve the problem soon from any of the sides.

NazerkeT, 2020-06-04:
Here is the follow up for everybody struggling with the same issue. It turned out that, this segmentation error is due to the latest commit in Verilator. However, the new release is coming and the problem should be solved soon. For more details, please refer to https://github.com/verilator/verilator/issues/2391.

closed 2020-06-04 by NazerkeT