openhwfoundation/cva6 #436: Cannot use verilator to generate waveform (vcd file)

ckj119940887, 2020-05-31:
I'd like to use verilator to generate waveform (.vcd file).
I used the command "make verilate DEBUG=1" and generated Variane_testharness.
Then, I executed the "work-ver/Variane_testharness --vcd=ckj.vcd rv64um-v-divuw".
As a result, it didn't generate the vcd file.

zarubaf, 2020-06-15:
I've just tried it on my side with: `Verilator 4.035 devel rev v4.034-100-g01f3e81a` and it generates the  VCD file.
Are you sure you don't have any errors before? Can you try deleting the `work-ver` folder and re-issuing the instructions? I assume you have a valid binary (called `rv64um-v-divuw`) at the place you are issuing the verilator command.

spidugu444, 2022-07-29:
Hi,

I am also trying to generate waveform in verilator for cva6 in core-v-verif environment.
 please let me know if this issue got resolved and  suggest how to enable  generation of vcd.

MikeOpenHWGroup, 2023-02-17:
Hi @ckj119940887, thanks for your interest in CVA6.  This issue has not been updated in more than six months, so I will assume it is resolved and will close this issue.  If you are still having trouble, please feel free to open another.

@spidugu444, if you are having trouble with the core-v-verif environment, please open an issue in that repo. Thanks!

closed 2023-02-17 by MikeOpenHWGroup