openhwfoundation/cva6 #439: make verilate  failed

mikemusasa, 2020-06-01:
Hi ! 
I am trying to run make verilator but i am getting this error . Which i do not fullty understand . any help ?  

error  
Makefile:399: recipe for target 'verilate' failed
make: *** [verilate] Error 1

zarubaf, 2020-06-15:
Unfortunately, I can't help you with this kind of error report, it is very generic. Do you have `verilator` installed?

mikemusasa, 2020-06-23:
Hi, I do have a verilator installed that works since it's helped me build for another core.    I tried to reinstall the arianne core and  i am now getting this error message  that it cannot find  <fesvr/dtm.h>  but i dowloaded the riscv-pk from risc tools  and in there , there is no fesvr/dtm as well.  Where can i find it to be able to build verilate ?

MikeOpenHWGroup, 2023-02-17:
Hi @mikemusasa, thanks for your interest in CVA6.  This issue has not been updated in more than two years, so I will assume it is resolved and will close this issue.  If you are still having trouble, please feel free to open another.

closed 2023-02-17 by MikeOpenHWGroup