openhwfoundation/cva6 #440: Need to enable FD instructions in spike

suhasskrishanmurthy, 2020-06-02:
Hi, 
 I am trying to enable FD extension on spike to test for tandem. Currently it is running for rv64imac. I checked the files , FD instruction is present. But they are not linked in spike executable present in install directory of tb/risc-isa-sim/install/bin. why is that?
Please can anyone help.

zarubaf, 2020-06-15:
The `tandem` simulation is not a supported feature at the moment. I've used it for bringing-up Linux which does not use `FD` in its early stage.

closed 2020-06-15 by zarubaf

chenc6, 2022-05-22:
Hi @zarubaf 
Seems like the current version of tandem does not support "FD". Is there a way to enable it?

Thank you

zarubaf, 2022-05-23:
Unfortunately, `tandem` is still not an officially supported feature. Of course I am happy if people get it to work i.e., issue pull requests for it :-)