openhwfoundation/cva6 #442: Changing PTE permission crashes kernel v5.7

atishp04, 2020-06-04:
The following commit in 5.7-rc1 added STRICT_KERNEL_RWX support and broke Linux for openpiton/ariane core.

commit d27c3c90817e
Author: Zong Li <zong.li@sifive.com>
Date:   Tue Mar 10 00:55:41 2020 +0800

    riscv: add STRICT_KERNEL_RWX support
   
    The commit contains that make text section as non-writable, rodata
    section as read-only, and data section as non-executable.
   
    The init section should be changed to non-executable.
   
    Signed-off-by: Zong Li <zong.li@sifive.com>
    Signed-off-by: Palmer Dabbelt <palmerdabbelt@google.com>

Here is a hack to make it work (tested with master OpenSBI)

atish@jedi-01:~/workspace/linux$ git diff
diff --git a/arch/riscv/Kconfig b/arch/riscv/Kconfig
index 62f7bfeb709e..f176c29d4f27 100644
--- a/arch/riscv/Kconfig
+++ b/arch/riscv/Kconfig
@@ -60,7 +60,6 @@ config RISCV
        select ARCH_HAS_GIGANTIC_PAGE
        select ARCH_HAS_SET_DIRECT_MAP
        select ARCH_HAS_SET_MEMORY
-       select ARCH_HAS_STRICT_KERNEL_RWX
        select ARCH_WANT_HUGE_PMD_SHARE if 64BIT
        select SPARSEMEM_STATIC if 32BIT
        select ARCH_WANT_DEFAULT_TOPDOWN_MMAP_LAYOUT if MMU
diff --git a/arch/riscv/mm/pageattr.c b/arch/riscv/mm/pageattr.c
index 728759eb530a..7f8a2386af8c 100644
--- a/arch/riscv/mm/pageattr.c
+++ b/arch/riscv/mm/pageattr.c
@@ -116,10 +116,9 @@ static int __set_memory(unsigned long addr, int numpages, pgprot_t set_mask,
 
        if (!numpages)
                return 0;

        down_read(&init_mm.mmap_sem);
-       ret =  walk_page_range_novma(&init_mm, start, end, &pageattr_ops, NULL,
-                                    &masks);
+       //ret =  walk_page_range_novma(&init_mm, start, end, &pageattr_ops, NULL,
+       //                           &masks);
        up_read(&init_mm.mmap_sem);
 
        flush_tlb_kernel_range(start, end);

Here is my theory:
The above commented line changes the kernel init text+data pages to non-executable.
Kernel tries to flush all pages after that and makes a SBI call for sfence.vma. Once OpenSBI flushes
the entire TLB for all cores, it returns to the kernel. But kernel never resumes. My assumption is that
changing PTE entry attributes is not behaving correctly in ariane core. I would be happy to proven wrong and debug more in kernel and/or OpenSBI. However, I have spent couple of days analyzing the issue and this seems to most plausible theory till now.

zhaoxiahust, 2020-07-01:
Is the environment single-core or multi-core?

MikeOpenHWGroup, 2023-02-17:
Hi @atishp04, thanks for your interest in CVA6.  This issue has not been updated in more than 2.5 years, so I will assume it is resolved and will close this issue.  If you are still having trouble, please feel free to open another.

closed 2023-02-17 by MikeOpenHWGroup