openhwfoundation/cva6 #444: baremetal applications on ARIANE

sjthales, 2020-06-09:
Hi,

I am trying to run the command below:

`make sim elf-bin=$RISCV/riscv64-unknown-elf/bin/pk target-options=hello.elf`

There is nothing on the tx signal of uart, I delved into the waves to understand.
At a moment the uart_putchar fonction of the proxy kernel was executed but there was no activity in uart peripheral.
Indeed, when I looked at the PK codes, I didn't see link between the architecture and PK codes.
Especially, I didn't see the definition of the uart peripheral address.

Is there a BSP or HAL which allows executing baremetal application on ariane?

Regards

SJ

zarubaf, 2020-06-15:
The `pk` uses a memory mapped location to perform syscalls (e.g., `write`).  The `riscv-fesvr` reads/writes the memory location and does the printing on the host system. 

Unfortunately, right now we do not provide a BSP or HAL.

sjthales, 2020-06-15:
Thank a lot for your answer.

closed 2020-08-24 by zarubaf