openhwfoundation/cva6 #446: How to determine whether the program is running normally？

Yi1101, 2020-06-15:
Dear Dr.  Zaruba,
when running the following command:

> make sim elf-bin=$RISCV/riscv64-unknown-elf/bin/pk target-options=hello.elf

```
The Questa Sim Transcript shows:
# Hello Ariane!\nUVM_INFO tb/ariane_tb.sv(113) @ 101218950: reporter [Core Test] *** SUCCESS *** (tohost = 0)
#  Note: $finish    : tb/ariane_tb.sv(116)
#    Time: 101218950 ns  Iteration: 3  Instance: /ariane_tb
```

It seems that it has run successfully, but which module should I check for whether it  run in the right step. I have tried to look into the SRAM ( the one supports the FPGA), and found it has some values. However, it does not have any values until nearly done, so I don't know when the hello.elf begins to be executed and what time does it finish. Hence, the problem is how to check the behavioral simulation is correct ？

zarubaf, 2020-07-01:
The end of computation is set by the RISC-V front-end server. It is a magic memory location called `.tohost` in your ELF file, so the exact location varies from binary to binary. I guess if you want to have precise control you would need to provide your own software/hardware framework which agrees on a fixed location.

closed 2020-07-01 by zarubaf

Yi1101, 2020-07-07:
So that why you use the Debug Module Interface (DMI) to communicate with the DUT, what's more, you are using the Front-End Server (FESVR) to facilitate communication between the host and the DUT ?