openhwfoundation/cva6 #447: SRAM area accounts for 94% of the total area

Yi1101, 2020-06-16:
Dear Author,
When implemented the Ariane core in Silicon, the DC result shows that the SRAM area accounts for 94% of the total area. The synthesis time takes more than 20 hours. The SRAM I used is the one in:src/fpga-support/rtl/SyncSpRamBeNx64.sv. 
By the way, there are two sram.v file. one is in: src/util/sram.sv (FPGA support SRAM) and the another one is in: common_cells/src/sram.sv (SRAM Behavioral Model). Could I use the common_cells/src/sram.sv to replace the src/util/sram.sv ? No matter which method, they all use the registers to build the logic and all take a long time in synthesis? 
That's the reason why you use the std cell to build the SRAM as your thesis mentioned???

zarubaf, 2020-06-16:
The SRAMs you mention are behavior only models. For an implementation in your target technology you want to replace those with generated ones for your technology. My advice would be to wrap them in a similar interface as the behavioral SRAMs so that you don’t need to change the instantiation.

Yi1101, 2020-06-16:
Many thanks for your reply. As you mentioned in the thesis, which used the memory compiler to build the SRAM. However, as i known, this will take 3 cycles delay, but the behavioral SRAM takes only 1 cycle. But i can't see the place to set the parameter on the cycle delay. Do I understand the SRAM cycle in the right way?

zarubaf, 2020-06-16:
Unfortunately, you need relatively fast SRAMs with a cycle of delay (as in the behavioral SRAM).
All the technologies I've been working with provide such SRAM macros (sometimes they are called register files). Albeit I've never used it myself, maybe [OpenRam](https://github.com/VLSIDA/OpenRAM) is something that could work for you?

Yi1101, 2020-06-16:
Really thanks for your advice, nice and in time advice, but I will use another way to get the right SRAM. Happy every day!

closed 2020-06-17 by Yi1101