openhwfoundation/cva6 #448: Co-simulation mismatch when reading stval CSR

kabylkas, 2020-06-17:
### To reproduce
This failure happens when running one of the riscv-isa tests: `rv64ui-v-bltu`.

```c
make verilate DROMAJO=1
make run-dromajo-verilator BIN=$(RISCV)/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-v-bltu
```

### Details
According to ISA spec (from what I understood, I might be wrong), `stval` should be set to `0` unless core trapped because one of the following:
* hardware breakpoint
* instruction-fetch fault
* load or store address-misaligned fault
* access or page-fault exception
* illegal instruction trap

However, Ariane seems to set the value of `stval` when trapping due to `ecall` (which is not one of the above). It sets stval to `0x73` which is the machine code for ecall instruction. I suspect same maybe true for `mtval` csr.

Here is Dromajo trace (read comments):
```c
// commit hundreds of instructions
0 0x0000000000002e58 (0x00301a63)                        bne     zero, gp, pc + 20 
0 0x0000000000002e6c (0x00100513) x10 0x0000000000000001 li      a0, 1
0 0x0000000000002e70 (0x00000073)                        ecall
1 0xffffffffffe000c4 (0x14011173) x2  0xffffffffffe096d0 csrrw   sp, sscratch, sp <-- trap due to ecall
1 0xffffffffffe000c8 (0x00113423)                        sd      ra, 8(sp)
1 0xffffffffffe000cc (0x00313c23)                        sd      gp, 24(sp)
// commit more instructions
1 0xffffffffffe00148 (0x100022f3) x5  0x8000000200006000 csrr    t0, sstatus
1 0xffffffffffe0014c (0x10513023)                        sd      t0, 256(sp)
1 0xffffffffffe00150 (0x141022f3) x5  0x0000000000002e70 csrr    t0, sepc 
1 0xffffffffffe00154 (0x10513423)                        sd      t0, 264(sp)
1 0xffffffffffe00158 (0x143022f3) x5  0x0000000000000000 csrr    t0, stval // <-- reading stval
[error] EMU PC ffffffffffe00158, DUT PC ffffffffffe00158
[error] EMU INSN 143022f3, DUT INSN 143022f3
[error] EMU WDATA 0000000000000000, DUT WDATA 0000000000000073 << mismatch
[error] EMU MSTATUS 8000000a00006000, DUT MSTATUS 00000000
[error] DUT pending exception 8 pending interrupt -1

```

zarubaf, 2020-07-01:
Re-reading the section in the ISA manual I got the same impression as you. Thanks for pointing that out. I'll flag it as a bug.

MikeOpenHWGroup, 2023-02-17:
Hi @zarubaf, this issue is rather old, but you've acknowledged it as a bug.  Can we get it fixed?