openhwfoundation/cva6 #450: Direction Change of file "registers.svh"

Yi1101, 2020-06-18:
Dear Dr.  Zaruba,

May be you will not take charge of the Issues, but really miss you and your reply! But there is no need to reply this issue. I just want to list it out for avoiding the same error, when someone runs it.

when run simulation, some errors occur, because the change direction of file "registers.sh".
Before it was in:
`common_cells/registers.svh`
Now it is in:
`common_cells/include/common_cells/registers.svh`
Hence the change the code:
`include "common_cells/registers.svh`
into 
`common_cells/include/common_cells/registers.svh`
The listed are the related files
```
fpnew/src/fpnew_cast_multi.sv
fpnew/src/fpnew_divsqrt_multi.sv
fpnew/src/fpnew_fma.sv
fpnew/src/fpnew_fma_multi.sv
fpnew/src/fpnew_noncomp.sv
fpnew/src/fpnew_opgroup_multifmt_slice.sv

```

zarubaf, 2020-06-19:
I am still helping on issues here ;-)

The two files should be identical so the idea is to point your include path to `common_cells/include/`. So the ``include "common_cells/registers.svh"` will be found.

Yi1101, 2020-06-19:
Wonderful U & Nice weekend !

closed 2020-06-19 by Yi1101

CVYang, 2025-07-07:
hello, i change the direction as shown,but error occuerd while compiling, the error info is "no such file or directory " .
compiler : vcs