openhwfoundation/cva6 #451: Request for contribution

NazerkeT, 2020-06-20:
Dear Florian,
Congratulations with passing Ariane to Open HW Group!
Wish Ariane all the best!

I am motivated by the idea behind open source development and PULP project, so want to contribute to Ariane as a summer project. I want to start my journey by adding support for Bit Manipulation Extension. Nevertheless, first, I want to understand whether is it really necessary and useful for Ariane, or is it something extra that you don't want to consider for now? If B extension is something wanted, then is it possible to make some plan on contribution as for A extension support to make things clear, or could I be addressed to someone interested in this?) 

On the other hand, if B extension is not necessary for the present, then I am open to any new idea on contributions as a project. For example, is adding support for out of order executions interesting for Ariane?

P.S. Basically, I have written an email to you too, but I am in worries whether is it got into spam or not. I am sorry if I am writing too much)

zarubaf, 2020-06-23:
Hi @NazerkeT,

thanks for your interest in contributing to Ariane and the hardware ecosystem in general. I think the B extension would make an excellent contribution to Ariane/CVA6 and to RISC-V. 

Another interesting feature could be an optional L2 TLB or having an optional SV48 address translation mode.

Let me know what you think and where you would be excited to contribute?

Cheers,
Florian

NazerkeT, 2020-06-24:
Hi @zarubaf,
All the options are exciting and appealing to me :). However, as long as I have already started to study RISCV Bit Manipulation, I think I better, first, go with adding support for the B extension. For now, I have completed one cycle implementation for `pcnt, clz, ctz`(some debug is needed for `ctz`). I've implemented them as functions, as I was initially planning for possible reusability of above-mentioned instructions. However, I rarely saw function implementations in ALU designs (after searching over some RISCV core repos including Ariane). Should I avoid implementation using functions? 

One more question, before processing to other instructions and considering possible architecture changes, could I be shared with what instructions are desired to be implemented in Ariane? And are there any other requirements on extension support to follow as of specification/technical task? 

Best regards,

NazerkeT, 2020-06-29:
Hi @zarubaf !
Here are some updates on what I have done on my way to B extension support. For now, I have completed rtl implementation of basic bit manipulation instructions (count leading/trailing zeros, count bits set, logic-with-negate, pack, min/max, sign extend, single bit, shift one's). So, now, I am moving towards a bit permutation, ternary bit instructions in line with bit extension draft doc. I will also look at Ariane architecture source files once again, this time far more carefully, to search for how I can adapt implemented instructions and to check whether I am doing things right or not. 

I will let you know some updates by the next Monday or maybe earlier. Hope to make some discussions soon :). That being said, I guess you are extremely busy these times too, my sincere apologies for interrupting you. 

_Best regards,
Nazerke_

NazerkeT, 2021-03-27:
Hi, there!
I recently recall that I have forgotten to put the references for what I have done on bit extension support during Summer 2020. So here I am :)

**[What is done]**
You can refer to https://github.com/NazerkeT/ArianeDemo for the implementation details. In general, this RISCV Bit Manipulation extension is addable to Ariane and built for `Zbb, Zbe, Zbp, Zbs` subgroups. Hardware behaves well for the basic(brute force) test of all instructions in the above-mentioned groups, but not yet integrated with the Ariane Decoder. In general, the approximate size after implementing RVB support is twice the size of the original ALU+MULT according to Vivado synthesis. 

Here `Zbb` denotes - basic bit instructions, `Zbe` - extract/ deposit instructions, `Zbp` - bit permutation instructions, `Zbs` - single bit instructions.

**[What is not finished]**
Actually, I have also tried to configure the decoder to test the core completely with the compiler support and Dhrystone benchmark in the OpenPiton environment. Nevertheless, I had some issues with the decoder integration and time constraints with studies also. So, one may continue this if wants :). As for my understanding compilers are now more prepared and mature to the RISC-V instructions than a year ago, but for more information please refer to riscv-bit-extension and gcc Github pages. Referring to lowRISC/ibex will also be extremely useful. They have already implemented it. 

**[Resources]**
As part of the bit extension implementation, I have referred to the RISC-V Bit Manipulation Extension Manual and Y. Hilewitz, Ruby Lee papers most of the time. However, Bit Extension Implementation in Ibex RISC-V Core was also very helpful to rely on. Here are the useful resources if someone else is interested in building the bit extension support to some other RISCV cores or to CVA also. 

RISCV-Bitmanip page: https://github.com/riscv/riscv-bitmanip

Ibex RISC-V Core: https://github.com/lowRISC/ibex

Hilewitz Y., Lee R., 2008.  "Fast Bit Gather, Bit Scatter and Bit Permutation Instructions for Commodity Microprocessors". _J Sign Process Syst (2008) 53:145–169._ DOI 10.1007/s11265-008-0212-8

Henry S. Warren. 2012. "Hackers Delight, Second Edition"

**Last but not least, huge thanks to @Jbalkind  for mentoring!**

M-Ijaz-10x, 2022-03-04:
I am working on completing this task of adding support for bitmanip Zba, Zbb, Zbc, and Zbs extensions to the CVA6.
I want to thank @NazerkeT for her contribution, which gave me a jump start on this task.
I have started working on it and expect to complete it by the end of April 2022, along with proper documentation, verification, and synthesis results. cc @zarubaf

NazerkeT, 2022-03-05:
Wow, glad to hear @M-Ijaz-10x! Good luck!

fatimasaleem, 2022-04-06:
Hi @Iripi97 
The fu_op enum contains the self-defined operations to mux through the different operations that one of the functional units can perform. This is NOT reference to the 7-bit instruction opcode defined in RISC-V.

MikeOpenHWGroup, 2023-02-17:
Hello @NazerkeT, thanks for initiating this!  As per @fatimasaleem's last comment, the bitmanip extension is "in", so I will close this issue.

Congratulations all - this is a BIG deal.

closed 2023-02-17 by MikeOpenHWGroup