openhwfoundation/cva6 #454: The Gate-level Simulation

Yi1101, 2020-06-20:
Dear Dr. Zaruba,
Really sorry for making too much trouble. This time I really have no idea. When running the gate-level simulation, the steps and problems as the followed.
Steps:

> 1. Writing a new bash file to run the gate-level simulation, the bash file followed some commands from your Makefile
> 2. Adding the netlist of ariane core (without sram, which has been pragmaed when run DC) 
> 3. Adding other needed modules and submodules in ariane_testharness.sv
> 4. The sram use the behavioral one
> 5. Running the bash file and the elf file is the one provided in the front page` "Hello Ariane"`
> 6. Waiting and waiting, until now, it already took more 24 hours, but the QuestaSim still runing.

My problems are:

> 1. Every time run the QuestaSim, tbh, I don't know which module and signal should I check？？？ I just waiting the result print on the Transcipt. If it shows "Hello Ariane" and then I know it works.
> 2. As I know, the elf file needs to be add into SRAM beforehand, that is the way of bevioral SRAM, but when using the SRAM macros, how to add elf into SRAM macros？？？   
> 3. The default setting of branch prediction in ariane_pkg is:` RASDepth: 2,BTBEntries: 32,BHTEntries: 128.` When set the parameter as your mentioned in the thesis: `RASDepth: 0, BTBEntries: 8, BHTEntries: 8`. Then the RAS module will has error, for the overflow of the signal, so in my opinion, when set  `RASDepth: 0`, I should delete all the RAS related module in Ariane, am I right???

Really looking forward to your reply!!!

zarubaf, 2020-06-23:
> 1. Every time run the QuestaSim, tbh, I don't know which module and signal should I check？？？ I just waiting the result print on the Transcipt. If it shows "Hello Ariane" and then I know it works.
> 2. [...]
If you use the test harness from this repo the core should preload its own memories. I think the main problem you will run into is that you can't mix the behavioral code with the gate-level code as you will run into tons of delta cycle issues. We don't really provide a good strategy for post-layout simulation here. What you would normally want to do is to build a testbench which applies the signals after a safe time frame (i.e., 20% of clock cycle) and samples them in another safe time frame (i.e, 80% of the clock cycle).
PLS, in general, will most likely be too slow to really get a lot of software to run on it.
> 3. The default setting of branch prediction in ariane_pkg is:` RASDepth: 2,BTBEntries: 32,BHTEntries: 128.` When set the parameter as your mentioned in the thesis: `RASDepth: 0, BTBEntries: 8, [...]
RasDepth in the thesis meant there wasn't a RAS in the version used in the paper. But good point, we would need to handle some more graceful handling (i.e., not including the RAS if it is set to 0).

Yi1101, 2020-06-24:
Thanks for your reply! I felt sorry that I had used triple "?", until today I realize it means impolite. Sorry again!

closed 2020-06-24 by Yi1101