openhwfoundation/cva6 #457: where is the online documentation?

bp-ouhaha, 2020-06-22:
Hi all,

I've been looking for the online documentation of CVA6 core. Can't find it. I was hoping to find something nice like lowRISC/ibex core, with the protocols and integration example.

I've looked into docs folder, but that seems to be source-file you have to compile/build yourself.

Thanks in advance.

zarubaf, 2020-06-23:
I think the original (outdated) online documentation got lost in the transition of the core.  Documentation will be added (similar to Ibex) in the very near-term future.

closed 2020-06-23 by zarubaf

whutddk, 2020-06-28:
@bp-ouhaha 

Hi, I'm searching for the docs, too. And I found that the source files of documentation were in 

> docs/_docs/arch

The graphs were in 

 > docs/img

Hopes it useful for you.