openhwfoundation/cva6 #458: Error during ariane-verilator simulation

mervinj, 2020-06-24:
I have cloned the latest commit and simulated the helloworld program with PK. It gives the following error during 

$work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk hello.elf

Variane_testharness: ../fesvr/dtm.cc:612: void dtm_t::tick(bool, bool, dtm_t::resp): Assertion `resp_wait' failed.

KIndly help to resolve the issue

Yi1101, 2020-06-24:
Please check your environment carefully, check the RISCV &  PATH, and follow the Getting Started strictly.

jimmysitu, 2020-09-25:
Hi, @mervinj 
I got the same error on one of my ubuntu 18.04 machine.
But is work fine when I change to other machine.

I think it might be a conner issue with verilator or system package, any idea?

mervinj, 2020-09-25:
Hi jimmysitu,
Did You solve the issue by trying in different ubuntu version or same version diff. machine. Kindly share the script if the problem is resolved.

MikeOpenHWGroup, 2023-02-17:
Thanks for your input on this issue @jimmysitu.

Hi @mervinj, thanks for your interest in CVA6.  This issue has not been updated in more than 2.5 years, so I will assume it is resolved and will close this issue.  If you are still having trouble, please feel free to open another.

closed 2023-02-17 by MikeOpenHWGroup