openhwfoundation/cva6 #460: Is it possible to dump wave traces to fst instead of vcd?

NazerkeT, 2020-06-28:
Hi there,
Coming back to simulations again, this time, I wanted to dump waveforms to `fst` instead of `vcd` during simulation readily to save memory. This is because of previous simulation results leading up to 20GB dumped vcd size. That was for simple `c=a+b` C source file. However, after converting that vcd file to `fst` with `vcd2fst`, filesize surprisingly reduced to 57.6MB. 

So, first of all, I tried to replace `--trace` with `--trace-fst` in Ariane Makefile according to Verilator manual and also tried to change `ariane_tb.cpp` for `VerilatedFstC` objects and `#include headers`, but things went unsuccessful. I think I am a bit confused and missing some important points. Could you help me with this extent? By the way, I have also tried to refer to [related RocketChip issue](https://github.com/chipsalliance/rocket-chip/issues/1689), however, that went unsuccessful too.

zarubaf, 2020-07-01:
I personally don't have any experience with the FST dumping. You may want to ask on the Verilator project?

NazerkeT, 2020-07-01:
> I personally don't have any experience with the FST dumping. You may want to ask on the Verilator project?

Okay, I will ask there too.

RaphaelKlink, 2020-07-01:
You can go a different way.
The gtkwave package has a built in vcd2fst function:
https://linux.die.net/man/1/vcd2fst

You can create the vcd traces and then convert them to fst

NazerkeT, 2020-07-01:
> You can go a different way.
> The gtkwave package has a built in vcd2fst function:
> https://linux.die.net/man/1/vcd2fst
> 
> You can create the vcd traces and then convert them to fst

Thanks for the help. That command is really helpful and I have already tried it. However, for this time, just wanted to see whether is it possible to dump readily to fst. Do you have any ideas on?

RaphaelKlink, 2020-07-01:
no i do not think so. 
When i used fst before it only saved space in storage.
When i opened a fst file with gtkwave it was decoded to the original size from the original vcd file.
This could indicate, that fst can only be used with a complete set of data and verilator could not create a fst file as it would need to change it after every dump step.

That is a very good space saving. When i used vcdtofst for a different project i was able to reduce 80 GB files to 15 GB files.

RaphaelKlink, 2020-07-01:
Or do you need a different class

in the header file

// VerilatedFstC
/// Create a FST dump file in C standalone (no SystemC) simulations.
/// Also derived for use in SystemC simulations.

it is stated that it this version can only be used for C only. Does verilator uses SystemC ?

NazerkeT, 2020-07-01:
> No not really
> When i used fst before it only saved space in storage.
> When i opened a fst file with gtkwave it was decoded to the original size from the original vcd file.
> This could indicate, that fst can only be used with a complete set of data and verilator could not create a fst file as it would need to change it after every dump step.
> 
> That is a very good space saving. When i used vcdtofst for a different project i was able to reduce 80 GB files to 15 GB files.

According to Verilator issues from 2019 spring onwards, it is possible to dump to fst files instead of vcd. Maybe it can be helpful for you too. Regarding SystemC, I am simulating C files, not SystemC. All Ariane testbenches were built in C, C++ too, verilated core model is itself in C++. If I am understanding things right, it should be possible to dump to fst, but for now, I dont know how, my trials went unsuccessful.

xushengj, 2021-03-08:
We save fst trace by:
1. Create a named pipe (`mkfifo trace.vcd`)
2. At the downstream of the pipe, use vcd2fst to do the conversion on-the-fly in background (`vcd2fst -v trace.vcd -f trace.fst &`)
3. Run the verilator model as usual and set the vcd trace output to the named pipe
Hope this works for you.

NazerkeT, 2021-03-09:
> We save fst trace by:
>
>     1. Create a named pipe (`mkfifo trace.vcd`)
>
>     2. At the downstream of the pipe, use vcd2fst to do the conversion on-the-fly in background (`vcd2fst -v trace.vcd -f [...]

I will try this out, thanks!

MikeOpenHWGroup, 2023-02-17:
By now everyone probably knows that Verilator can dump `fst` files, and that `corev_apu/tb/ariane_tb.cpp` has been updated to do this as of `871be7c79` (on 2022-09-27).  Closing.

closed 2023-02-17 by MikeOpenHWGroup