openhwfoundation/cva6 #462: Simulation Time is Different in QuestaSim

Yi1101, 2020-06-30:
Dear Dr. Zaruba,

When I tried to use my own bash file to run Behavioral Simulation with QuestaSim, a strange phenomenon appeared. The simulation time of my script (245 264 250 ns) was more than twice longer than your Makefile ( 101 218 950 ns). 

Script_1. The vsim command from your Makefile is:
`vsim "+permissive" -t 1ns -coverage -classdebug -do " log -r /*; run -all;" "+jtag_rbb_enable=0" -lib work "+MAX_CYCLES=10000000" "+UVM_TESTNAME=core_test" "+BASEDIR=tmp/riscv-tests/build/isa/" "+UVM_NO_RELNOTES" "+UVM_VERBOSITY=LOW" -gblso /home/riscv-gnu-toolchain/lib/libfesvr.so -sv_lib work-dpi/ariane_dpi ariane_tb_optimized "+permissive-off" "++/home/riscv-gnu-toolchain/riscv64-unknown-elf/bin/pk" "++hello.elf" `

Script_2. The vsim command from my own bash file is:

`vsim "+permissive" -t 1ns -coverage -classdebug -suppress 7063 -suppress 7061 -suppress 2732 -suppress 3009 -do " log -r /*; run -all; " -lib work "+MAX_CYCLES=10000000" "+UVM_TESTNAME=core_test" "+BASEDIR=tmp/riscv-tests/build/isa/" "+UVM_NO_RELNOTES" "+UVM_VERBOSITY=LOW" -gblso /home/riscv-gnu-toolchain/lib/libfesvr.so -sv_lib work-dpi/ariane_dpi ariane_tb "+permissive-off" "++/home/riscv-gnu-toolchain/riscv64-unknown-elf/bin/pk" "++../../../ariane/hello.elf"`

Script_3. Your Makefile which runs in preload mode (Command: make sim preload=hello.elf)

`vsim "+permissive" -t 1ns -coverage -classdebug -do " log -r /*; run -all;" "+PRELOAD=hello.elf" "+jtag_rbb_enable=0" -lib work "+MAX_CYCLES=10000000" "+UVM_TESTNAME=core_test" "+BASEDIR=tmp/riscv-tests/build/isa/" "+UVM_NO_RELNOTES" "+UVM_VERBOSITY=LOW" -gblso /home/riscv-gnu-toolchain/lib/libfesvr.so -sv_lib work-dpi/ariane_dpi ariane_tb_optimized "+permissive-off" "++none" "++" 
`

The questions are:

> 1. Even though Script_1 and Script_2 is almost the same when running vsim, but why they have a great difference in behavioral simulation time?
> 2. It seems that preload memory has not effect on reducing total simulation time, when compare the simulation time result with Script_1 and Script_2, the Script_3  take the most time.  So why?
> 3. What may caused the shorter simulation time in Script_1?

zarubaf, 2020-07-01:
The preload should technically be the fastest way, **but** most of the useful things (such as printf and end-of-computation detection) don't work anymore since `riscv-fesvr` is deactivated. As you don't see any difference in runtime and your program works I am assuming that it is actually didn't preload.

Can you point me to the differences in your commands?

Yi1101, 2020-07-01:
Thanks a thousand times. You and your reply light up my day. 

Before answering the question, I would like to give out my understand on "make sim" command. When running the "make sim" with QuestaSim, the whole flow could be divided into three steps:

> step_1.  vlog/vcom which compiles HDL code.
> step_2.  g++
> step_3.  vsim which run the QuestaSim simulation. Occupying most running time.

The running commands for:

> Script_1.  make sim elf-bin=$RISCV/riscv64-unknown-elf/bin/pk target-options=hello.elf 
> Script_2.  bash ../scripts/vsim.sh
> Script_3.  make sim preload=elf

The simulation time for:

> Script_1.  101 218 950 ns
> Script_2.  245 264 250 ns
> Script_3.  more than 1 541 788 010 ns
> Script_4. 101 233 230 ns

Phenomenon:

> 1. Script_1 and Script_2 are the same with step_1 and step_3, but have difference in step_2. But in my opinion, step_1 and step_2 don't take time. If step_3 is the same so that the time should be nearly the same. 
> 2. Script_1 and Script_3 are the same except step_3. Why step_3 could take such a long time?

My questions:

> 1. How to check the QuestaSim’s result, I have looking into the sram. Tbh, I am not sure about the value. Until now, my biggest problem always the one how to match hello.elf with SRAM (when it begins and ends)? (Not sure if the question is accurate)
> 2. Could you explain more about "since riscv-fesvr is deactivated", does it mean that it does not work  in Script_1 or it just don't exit ?
> 3. I would like to add some commands into my own script, these commands should have the function to printf xx% time of total simulation. I think you have already answered this question in another issue, looking into the .tohost in my ELF file. Is that a correct understanding?
> 4. As for why the preload didn't work, in which direction should I check?

What's more, I use a reverse way (based on the report from your Script_1) to get a new Script_4, which nearly has the same time with your Script_1. Fortunately, Script_4 is the one I am really looking for. Thank you for your effort again.

Yi1101, 2020-07-06:
I use the Netlist to replace the Ariane DUT. However, the QuestaSim has already run almost three days, the result has not yet come out. The Netlist simulation time has already reach the same as behavioral simulation time (Script_4 101 218 950 ns). I really get stuck.

closed 2020-08-24 by zarubaf