openhwfoundation/cva6 #464: Idea: Change register type detection

RaphaelKlink, 2020-07-01:
Hey,

currently, the Ariane core detects from the operation if the register is a scalar or a floating-point one.

I would suggest instead a register for this task. The type of register is saved in a 2-bit value with the three values:

Constant
Scalar
Floating-Point

This would reduce the register from 7-bits to 2-bits. 
This is already done in the scoreboard to speed-up the register arbitration. 
https://github.com/openhwgroup/cva6/blob/465bb209a02721cbaa6fa82a87df2eb464a39e60/src/scoreboard.sv#L70

What if this always done?

The decoder would be bigger but the register arbitration could be even faster. Only the type for the destination register is saved currently. With this change also the type for the source register would be saved. 
This would reduce the calculation if the destination and source register are of the same type. Only 4 bits then store all the necessary information. For FPGA this change is even better as most modern FPGA use LUTs with 4 or 6 inputs. This calculation can, therefore, be done with only one LUT.

The number of operations could even be reduced.  When the type of register is not saved in the operation itself. Different types of the same operation could be removed. For example, scalar add and floating-point add could be merged, as the Execution Unit is also determined by the decoder.

The integration of the constant type would also eliminate the need for the use_imm and use_zimm flags from the scoreboard entry.

This change would therefore only need 4 additional FFs per scoreboard entry.
2 FFs for the type of the destination register + 4 FFs for the two source registers - 2 FFs for the use_imm and use_zimm flag

When floating-point is not used the type could be changed to a 1-bit value:

Constant
Scalar

This meaning is then similar to the use_imm and use_zimm flags.

Would love to hear your opinions on this matter.

Best regards,
Raphael Klink

zarubaf, 2020-08-24:
That sounds like a good idea. A likely next step would probably be to quantify if this has a positive impact on timing/area.

MikeOpenHWGroup, 2023-02-17:
Hi @zarubaf.  This issue is more than 2.5 years old.  If there was a PR merged in to implement this idea, please close this issue.  Otherwise, let's either do it, or label it as `Wont-fix` and close the issue.

zarubaf, 2023-10-20:
Closing because of lack of interest and staleness.

closed 2023-10-20 by zarubaf