openhwfoundation/cva6 #468: Dhrystone and printf problem

epthales, 2020-07-07:
Hi,

I'm trying to run dhrystone on Ariane, to see the results on the terminal. However, when I try to do this command `work-ver/Variane_testharness tmp/riscv-tests/build/benchmarks/dhrystone.riscv` I get this response:
 `This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 38503
[396655] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396664] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396673] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396682] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396691] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396700] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396709] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396718] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396727] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396736] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396745] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396754] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396763] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396772] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396781] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396790] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396799] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396808] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396817] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396826] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396835] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396844] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396853] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396862] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396871] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396880] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396889] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396898] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396907] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396916] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396925] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396934] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396943] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396952] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396961] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396970] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396979] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396988] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[396997] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397006] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397015] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397024] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397033] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397042] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397051] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397060] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397069] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397078] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397087] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397096] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397105] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397114] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397123] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397132] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397141] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397150] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397159] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397168] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397177] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397186] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397195] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397204] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397213] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397222] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[397231] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[... 355 lines cut ...]
[400435] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[400444] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[400453] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[... cut ...]
[400598] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
[400607] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored

terminate called after throwing an instance of 'std::runtime_error'
  what():  bad syscall #2147505792`

With this command, we should have the dhrystone results on the terminal, but I don't know why there are "Responses Errors". I haven’t found the solution of this problem. Can you help me to solve it?

Yi1101, 2020-07-08:
Trying to add this `$RISCV/riscv64-unknown-elf/bin/pk` into your run command.

epthales, 2020-07-08:
> Trying to add this `$RISCV/riscv64-unknown-elf/bin/pk` into your run command.

Hello, I tried this command `work-ver/Variane_testharness $RISCV/riscv64-unknown-elf/bin/pk tmp/riscv-tests/build/benchmarks/dhrystone.riscv` but I get : 

This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 38983
terminate called after throwing an instance of 'std::runtime_error'
  what():  bad syscall #8319104478668415267

Yi1101, 2020-07-09:
Then try this pk, which is in the`/riscv-pk/build/pk`

epthales, 2020-07-09:
> Then try this pk, which is in the`/riscv-pk/build/pk`

I tried this pk and I get : [46886] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored
...
[2785492] %Warning: ariane_testharness.sv:720: Assertion failed in TOP.ariane_testharness.p_assert: B Response Errored

Yi1101, 2020-07-10:
Tbh, I don't think the command has any problems. Because, I had run all the benchmarks successfully. Hence, please make sure the environment you had built is full and accurate, which is really important.

zetalog, 2020-08-04:
Does ariane bench contains a hardware supporting PK magic call?

closed 2020-08-26 by epthales