openhwfoundation/cva6 #469: Physical Memory Map Configuration

Kendidi, 2020-07-08:
Hi,

I wonder where and how to configure physical addresses of hardware/module such as,
    1) DDR base address
    2) UART base address

for the Linux Kernel before building the image?  I plan to use OpenOCD and GDB to load the image.

Thanks a lot in advance!!
Ken

Yi1101, 2020-07-09:
In my opinion, in author's hardware code, which supports DMI protocal and the host is fesvr. So how could you use OpenOCD and GDB, which uses JTAG protocal? Please point it out if my understanding is not correct, many thanks!

Moschn, 2020-07-09:
@Kendidi Take a look at the device tree file ([ariane.dts](bootrom/ariane.dts)). It contains a description of the peripherals and to which address they are mapped. For example: the DDR base address is specified in line 29: `memory@80000000 {` wich means it is at 0x80000000. The UART is specified on line 60 `uart@10000000 {` and is at 0x10000000.

@Yi1101 As far as I undertstand, Ariane/cv6 implements the risc-v debug  spec. For JTAG specifically take a look at "6.1 JTAG Debug Transport Module". In practice this allows OpenOCD to talk to the core using the JTAG debug protocol. OpenOCD can then open a GDB session. fesvr also communicates with Ariane/cv6 according to the riscv-debug spec.

Kendidi, 2020-07-09:
> 
> 
> In my opinion, in author's hardware code, which supports DMI protocal and the host is fesvr. So how could you use OpenOCD and GDB, which uses JTAG protocal? Please point it out if my understanding is not correct, many thanks!

Kendidi, 2020-07-10:
I am currently trying to build **vmlinux** from **ariane-sdk** to load to FPGA (Modeled from Ariane) via OpenOCD and GDB, but with custom DDR (0x2_0000_0000) and UART (0x1A00_0000) base addresses.  Therefore I would appreciate if anyone can share how to modify the Linux kernel build so that the when Linux Kernel starts, it will know where the new physical addresses of the devices (e.g. DDR, UART...) are. Many thanks!

Moschn, 2020-07-13:
So to change that need to do two things:
- change the dts file and rebuild Ariane. The dts file is baked into ROM and will be presented to the bootloader and the kernel during boot.
- change the vmlinux to a different offset. I believe you can do that in buildroot/linux menuconfig by changing the linux defconfig file in the ariane-sdk. Not exactly sure how you would have to change riscv-pk to also move to a different base address. (You could also try to do this in the last step. `objcopy` accepts a parameter to relocate a binary. E.g., `riscv64-unknown-elf-objcopy -S -O binary --change-addresses -0x80000000 bbl bbl.bin`)

Kendidi, 2020-07-14:
@Moschn Thank you very much for your advice!  Since we do not use really Boot ROM and Bootloader at this point, can we modify the dts file(s) of the Linux Kernel instead?  Thanks.

Moschn, 2020-07-14:
The linux kernel does not have its own dts file. This file is usually supplied by the zero stage bootloader in ROM. Maybe there is a way to emulate this in GDB but I do not really know how. I guess you could put the dts file somewhere in memory and then put this address into the argument for vmlinux.

MikeOpenHWGroup, 2023-02-17:
Hi @Kendidi, thanks for your interest in CVA6.  This issue has not been updated in more than 2.5 years, so I will assume it is resolved and will close this issue.  If you are still having trouble, please feel free to open another.

closed 2023-02-17 by MikeOpenHWGroup

jimaandro, 2023-09-15:
I have a similar problem, my DDR base is 80_0000_0000 and ROM 8000_0000_0000. These are 64bit addresses, but the makefile in cva6-sdk accepts only 32bit address for UIMAGE_LOAD_ADDRESS and UIMAGE_ENTRY_POINT. So, instead of changing my design, is there any other way to produce the correct image?

rakeshnakkilla, 2026-08-17:
Hi @jimaandro ,

Is your issue resolved? I also have a similar issue.