openhwfoundation/cva6 #471: Update path to GitHub repo after move

mbrukman, 2020-07-09:
The "Getting Started" section of the [`README.md`][readme] currently says:

> Checkout the repository and initialize all submodules
>
> ```
> $ git clone https://github.com/pulp-platform/ariane.git
> [...]
> ```

However, that appears to be the old name and location of the repo, before it was moved to its current location (https://github.com/openhwgroup/cva6). While GitHub redirection enables this to continue to work<sup>1</sup>, this isn't technically broken, it is still confusing to readers as this implies they should be cloning another repo instead of this one.

It would be good to update it to point to the current location.

---

<sup>1</sup> at least until a new repo is created in the old location

[readme]: https://github.com/openhwgroup/cva6/blob/465bb209a02721cbaa6fa82a87df2eb464a39e60/README.md#getting-started

zarubaf, 2020-07-15:
We will happily accept a pull request with this contribution! Thank you very much!

MikeOpenHWGroup, 2023-02-17:
Hi @mbrukman, thanks for your interest in CVA6.  This issue has not been updated in more than 2.5 years, so I will assume it is resolved and will close this issue.  If the documentation is still incorrect, please feel free to open another, along with a pull-request with a suggested fix.

closed 2023-02-17 by MikeOpenHWGroup