openhwfoundation/cva6 #476: A corner case in std_icache when kill_s2 is asserted

timingwalker, 2020-07-16:
Hello,

We find a corner case associated with BranchPrediction/std_icache/PTW when we try to boot the Linux OS with std_cache_subsystem. An error address translation result is returned by PTW，which makes the OS crash.

### Phenomenon
When _std_icache_ send address translation requests to MMU continuously，if the following conditions are met：
1. A branch mispredict  occurs in frontend, which will assert _kill_s1_、_kill_s2_ signal 
2. FSM of _std_icache_（_state_q_ in std_icache.sv）happens to be inside **TAG_CMP** stage when _kill_s2_=1 
3. The killed request and the following normal request are TBL miss

In this situation，MMU receives a back-to-back request and ask PTW to handle them. It seems that MMU treat the back-to-back requests as a single request due to the long process time in PTW. The first request is not killed actually, and the result is returned by MMU as the second request. which will make the Linux OS crash.

### Detail 
when _std_icache_ FSM jump to **TAG_CMP**, it asserts _areq_o.fetch_req_ and sends a translation request to MMU: 
https://github.com/openhwgroup/cva6/blob/1b65b8865a505722b8bbbdd1a0e6dc959b28e166/src/cache_subsystem/std_icache.sv#L239
if _kill_s2_ is asserted at this state due to branch mispreditc in frontend:
https://github.com/openhwgroup/cva6/blob/1b65b8865a505722b8bbbdd1a0e6dc959b28e166/src/frontend/frontend.sv#L244
https://github.com/openhwgroup/cva6/blob/1b65b8865a505722b8bbbdd1a0e6dc959b28e166/src/frontend/frontend.sv#L247
FSM  in _std_icache_ will jump to **IDLE** to kill this request, but in fact this request has been sent to MMU: 
https://github.com/openhwgroup/cva6/blob/1b65b8865a505722b8bbbdd1a0e6dc959b28e166/src/cache_subsystem/std_icache.sv#L381

If the instruction  queue is not empty, _dreq_i.req_ from frontend will keep asserted, FSM in _std_icache_ jumps to **TAG_CMP** again and sends another translation request. MMU receives a back to back request from _std_icache_ like this: 

> _icache_areq_i.fetch_req_ :    XXXX______/¯¯¯\\___/¯¯¯\\______XXXX
(the first one is the killed request and the second one is a normal request)

The killed request makes the FSM in PTW jumping from **IDLE** to **WAIT_GRANT**. As there is no mechanism in PTW to cancel a translation，when the second request come, PTW can not feel it and do something(e.g. update the _ptw_pptr_q_)，it is still in the translation process of the first request. The result of the killed request is returned, which will be treated as the result of the second request by _std_icache_.

So the gap is : _std_icache_ kills the first request and sends another request, but PTW does't kill the first one!

### Solution
Maybe **TAG_CMP** can be added in this code to wait PTW exit safely : 
https://github.com/openhwgroup/cva6/blob/1b65b8865a505722b8bbbdd1a0e6dc959b28e166/src/cache_subsystem/std_icache.sv#L381

### Attachment
[image]

zarubaf, 2020-07-16:
Thanks a lot for the detailed description and for putting this together so nicely. It seems that the issue is also what #474 tries to resolve. I like the proposed solution from this issue better as it keeps the output signals stable and independent on the `kill_s2` signal. Only the next state logic is affected.

@masgia Your input would be very, very much appreciated if you can check whether the proposed fix of putting `TAG_CMP` in the list of states that should be waited does also solve this issue.

@timingwalker Can you confirm that your fix/solution solves the issue?

masgia, 2020-07-16:
That was actually my first try :)
But it resulted in a different crash, which I didn't want to debug since I already had an alternative ready.

I think my proposed solution is better, since it doesn't force the system to wait the translation of an address which has to be trashed anyway.

timingwalker, 2020-07-16:
I will confirm that sulution later @zarubaf . 
In fact, we have found another corner case in std_dcache which is associated with amo/ld instruction. I don't know if it is the same crash as @masgia. We will do a thorough analyses and report it later.

masgia, 2020-07-16:
@timingwalker the issue I saw looked exactly like the one you described here

timingwalker, 2020-07-23:
Putting TAG_CMP in the list of states that should be waited is not enough because FSM dumps to WAIT_ADDRESS_TRANSLATION, which makes it lost the kill_s2 signal.
TAG_CMP should jump to WAIT_ADDRESS_TRANSLATION_KILLED directly，i add these code after line 390 in std_icache.sv :
```
if (dreq_i.kill_s2 && (state_q==TAG_CMP) && !areq_i.fetch_valid)
    state_d= WAIT_ADDRESS_TRANSLATION_KILLED;
```

I meet another crash after this modify, but it is causeed by another issue #480 
After I fix #480 , this problem is resolved too.

MikeOpenHWGroup, 2023-02-17:
Hi @timingwalker.  Do you know the current status of this issue?