openhwfoundation/cva6 #477: Prefixing of module names and file names

Silabs-ArjanB, 2020-07-17:
In CV32E40P all module names and file names were prefixed with cv32e40p_ to avoid naming collisions at the chip level, see https://github.com/openhwgroup/cv32e40p/issues/276.

Can the same be done for all RTL files in CVA6 (I would propose a cva6_ prefix)?

In CV32E40P the following naming style is used:

- Filename : cv32e40p_\<something\>.sv
- Module name: cv32e40p_\<something\>
- Instantiation: cv32e40p_\<something\> () \<something\>_i

zarubaf, 2020-08-24:
We should probably do that. There is just the problem that that creates an un-reviewable amount of diff :/ So we probably need to tackle that step-by-step.

MikeOpenHWGroup, 2023-02-17:
> We should probably do that.

Agreed.  It is long past time that we did.

> There is just the problem that that creates an un-reviewable amount of diff :/ So we probably need to tackle that step-by-step.

A step-wise approach is best.  When can you take the first step?  :stuck_out_tongue_winking_eye: