openhwfoundation/cva6 #493: Errors when trying to run 'hello.elf' with Questasim

0ena, 2020-08-05:
Hi,

I am trying to use QuestaSim to run the 'hello.elf' but I get the following errors:

```
Error (suppressible): (vsim-3197) Load of "/eda/tools/riscv/riscv-tools/lib/libfesvr.so" failed: /eda/tools/riscv/riscv-tools/lib/libfesvr.so: cannot open shared object file: No such file or directory.
```
and 

`** Error: (vsim-7) Failed to open UCDB file "tmp/.ucdb" in write mode.`

I am attaching also the complete log:

[ArianeQuesta.log](https://github.com/openhwgroup/cva6/files/5029687/ArianeQuesta.log)

Anyone familiar with above errors?

Thank you in advance for your response.

Kind regards,
Nassos

0ena, 2020-08-06:
Hi I found this post:

https://www.gitmemory.com/issue/pulp-platform/ariane/390/597694633

I went on installing the riscv-fesvr version in the Ariane-sdk repository: https://github.com/riscv/riscv-fesvr/tree/8d108a0a647901550d95925549337c2c3aec9ac8

This created the requested 'libfesvr.so' and solved the first of the above errors.
However the second one `** Error: (vsim-7) Failed to open UCDB file "tmp/.ucdb" in write mode.` still remains. Even if I execute in su mode, I still get the same error.
Any ideas on how to solve it?

Also are there somewhere written the **versions** of all the tools that we should be using when working with Ariane?

Thanks,
Nassos

0ena, 2020-08-06:
I can see in the Makefile the following:

lines 216-231:

```
# look for testbenches
tbs := tb/ariane_tb.sv tb/ariane_testharness.sv
# RISCV asm tests and benchmark setup (used for CI)
# there is a definesd test-list with selected CI tests
riscv-test-dir            := tmp/riscv-tests/build/isa/
riscv-benchmarks-dir      := tmp/riscv-tests/build/benchmarks/
riscv-asm-tests-list      := ci/riscv-asm-tests.list
riscv-amo-tests-list      := ci/riscv-amo-tests.list
riscv-mul-tests-list      := ci/riscv-mul-tests.list
riscv-fp-tests-list       := ci/riscv-fp-tests.list
riscv-benchmarks-list     := ci/riscv-benchmarks.list
riscv-asm-tests           := $(shell xargs printf '\n%s' < $(riscv-asm-tests-list)  | cut -b 1-)
riscv-amo-tests           := $(shell xargs printf '\n%s' < $(riscv-amo-tests-list)  | cut -b 1-)
riscv-mul-tests           := $(shell xargs printf '\n%s' < $(riscv-mul-tests-list)  | cut -b 1-)
riscv-fp-tests            := $(shell xargs printf '\n%s' < $(riscv-fp-tests-list)   | cut -b 1-)
riscv-benchmarks          := $(shell xargs printf '\n%s' < $(riscv-benchmarks-list) | cut -b 1-)
```

lines 251-255:

```
# if defined, calls the questa targets in batch mode
ifdef batch-mode
    questa-flags += -c
    questa-cmd   := -do "coverage save -onexit tmp/$@.ucdb; run -a; quit -code [coverage attribute -name TESTSTATUS -concise]"
    questa-cmd   += -do " log -r /*; run -all;"
```

So from what I understand there should be inside the Ariane project a folder called `tmp`  or should I just change 'tmp' to the path where my `riscv-tests/build/isa` reside?

Thanks,
Nassos

zarubaf, 2020-08-24:
Yes, just create a `tmp` directory (or use execute `ci/make-tmp.sh` which does exactly that :D. You can then execute `ci/build-riscv-tests.sh` to build the tests in that tmp directory (that is what the Makefile expects).

closed 2021-04-01 by 0ena