openhwfoundation/cva6 #494: Make Verilate Error

liaolin1997, 2020-08-06:
Hi!
I'm trying to follow the steps of building the verilator model of ariane by using the makefile.I have installed the verilator and other riscv tools. However ,when I run the command"make verilate", I met some errors which are hard for me to understand.
The errors are as followings.Could you please give me some suggestion?

../tb/dpi/remote_bitbang.h: In constructor ‘remote_bitbang_t::remote_bitbang_t(uint16_t)’:
../tb/dpi/remote_bitbang.h:43:23: error: ‘remote_bitbang_t::recv_end’ will be initialized after [-Werror=reorder]
   ssize_t recv_start, recv_end;
                       ^~~~~~~~
../tb/dpi/remote_bitbang.h:29:7: error:   ‘int remote_bitbang_t::err’ [-Werror=reorder]
   int err;
       ^~~
../tb/dpi/remote_bitbang.cc:19:1: error:   when initialized here [-Werror=reorder]
 remote_bitbang_t::remote_bitbang_t(uint16_t port) :
 ^~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
Variane_testharness.mk:74: recipe for target 'remote_bitbang.o' failed
make[1]: *** [remote_bitbang.o] Error 1
make[1]: *** 正在等待未完成的任务....
make[1]: 离开目录“/home/sri/ariane/work-ver”
Makefile:469: recipe for target 'verilate' failed
make: *** [verilate] Error 2

Moschn, 2020-08-10:
Which version of verilator are you using? We generally require the newest version. This is **not** the version from the ubuntu repositories for example.

zarubaf, 2020-08-24:
Closing, feel free to re-open in case that didn't resolve the issue.

closed 2020-08-24 by zarubaf