openhwfoundation/cva6 #495: Debug with OpenOCD

DeltaEnvy, 2020-08-10:
Hi.
I have a problem about debug.

I use the I/O layout tool in Xilinx to specify JA1_P, JA1_N, JA2_P, JA2_N as TDO, TDI, TCK, and TMS respectively, and start OpenOCD according to your steps, and the following error will be reported.

root@delta:/home/delta/riscv-openocd# openocd -f fpga/ariane.cfg
Open On-Chip Debugger 0.10.0+dev-01229-g9ed670771 (2020-08-06-14:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : clock speed 1000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unsupported DTM version: 15
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet

Error: Unsupported DTM version: 15

Please tell me how to fix this problem.
Many thanks in advance.

Moschn, 2020-08-10:
It seems like you are neither using our release version, nor our xilinx toplevel and hhardware config. It will be hard to help you on your own design. I would suggest you take a look at how we did it in our FPGA mapping. See for example [fpga/constraints/genesys-2.xdc](fpga/constraints/genesys-2.xdc)

DeltaEnvy, 2020-08-11:
Oh,this is my mistake, I will try again and get back to you,thanks

zarubaf, 2020-08-24:
Closing, feel free to re-open in case it didn't work.

closed 2020-08-24 by zarubaf