openhwfoundation/cva6 #496: General question about memories in Ariane (cache and on-board DDR)

0ena, 2020-08-10:
Dear Dr. Zaruba (@zarubaf),

I have some questions regarding the memories in the Ariane Project (cache and on board DDR).
I am interested in building a system that will be able to run Linux on it. I would really appreciate it if you could take some time and provide me with an answer on the following:

a) Which parameters should I adjust in order to change the size of the Data Cache and Instruction Cache of Ariane? 
b) Is there a minimum ICache and DCache size that you suggest in order for Ariane to be able to support Linux?
c) Is there a minimum on-board DDR size suggested in order to be feasible to run Linux on Ariane?

Thank you in advance for your time and response.

Kind regards,
Nassos

zarubaf, 2020-08-24:
As configured now Linux requires a couple of MB (~64) main memory (DDR). You can adjust the number of ways to achieve a different size for both L1 instruction cache and data cache, be aware though that those cache configurations have likely not been tested. Technically there is no minimum cache size necessary to boot Linux (you can also boot it without any caches).

closed 2020-08-24 by zarubaf

0ena, 2020-08-24:
Thank you for your thorough answer Dr. Zaruba (@zarubaf), I really appreciate it!

As for a), can you please tell me which parameters adjust the sizes of ICache and DCache?

Thanks,
Nassos