openhwfoundation/cva6 #516: VCU118

jctullos, 2020-08-31:
I was looking to see if there was a fork or branch that has a VCU118 ddr4 implementation? I was reading one of the topics where someone used a VCU108, and had incorporated some of their build process. 

I was able to generate the bitstream/mcs, but for some reason I'm not getting any uart output after multiple tries. I figure I might have something wrong in the constraints file that I updated for the VCU118. 

Thank you!

RaphaelKlink, 2020-09-01:
Hey,

I did the VCU108 port. 

Yes, a wrong constraint file could be the problem. You could check this by debugging the uart signals in vivado. I could help you better with it if you posted the whole project including the new constraint file somewhere.

Did you used the built in uart from the board or an external one via a pmod header?
 I used the latter option. I could not get the built in uart to work on the VCU108.

Raphael

jctullos, 2020-09-01:
@RaphaelKlink 

Hey! Thanks for the help. I actually referenced your changes to help build the vcu118 one, it was a huge help.

So for the uart, I was actually able to get it working last night for the on board UART. So it prints out Hello World! But then doesn't initialize the SPI. I've got a logic analyzer hooked up to the PMOD pins and right now there's not even a clock coming out. I'm loading a new build to see if I fixed it, so we'll see.

I'll try and get the project uploaded and post a link. Maybe I've missed something. 

Thank you again for helping!

jctullos, 2020-09-01:
@RaphaelKlink 

Alright, all the changes should be uploaded to: https://github.com/jctullos/cva6

I believe I have all the VCU118 added in, let me know if I missed a file. 

FYI, for the VCU118 contraints, I'm using the VCU118 Rev 1.1 board which has BPI instead of the QSPI flash. So that's why it has BPI settings in it.

RaphaelKlink, 2020-09-01:
I look over your files. I only found one small mistake.

MISO is defined twice, but this should not be a problem.
set_property -dict {PACKAGE_PIN AW15 IOSTANDARD LVCMOS18} [get_ports spi_miso]
set_property -dict {PACKAGE_PIN AW15 IOSTANDARD  LVCMOS18} [get_ports spi_miso] ;PMOD0_2

Could you check the AXI into the SPI Interface?
Is there any activity?

Maybe it is not an SPI problem but a RAM problem. 
could you remove the ddr4 interface pins from the constraint file.
In my project vivado does this automatically from the chosen DDR Channel in the IP  Core. Maybe there was an error?

Where did your values for the Instantiation values for the IP Core for the DDR Channel came from?
Did you do them yourself or did vivado created them?

Could you send a screenshot from your UART console?

jctullos, 2020-09-08:
So I ended up getting the VCU118 build working for openpiton with ariane, so just using that now. 

Thanks again for the help!

closed 2020-09-08 by jctullos

michael-etzkorn, 2021-08-20:
I got that CONFIG.ADDN_UI_CLKOUT1_FREQ_HZ {200} wasn't a valid value for the DDR4 configuration you used. Thanks for putting down some legwork for getting cva6 compatible with vcu118.