openhwfoundation/cva6 #517: Program counter getting set to 0 eventually after a mispredict

JoshAbraham0, 2020-08-31:
Hello,

I am seeing an issue where `instruction_o.pc` will unexpectedly be set to 0 after `is_mispredict` is triggered. From what I see, `target_address` in `branch_unit.sv` gets set to 0, and triggers `is_mispredict` after being compared to the `branch_predict_i.predict_address`, which is 136. After `is_mispredict` is triggered, in `frontend.sv`, `npc_d` is set to `resolved_branch_i.target_address`, which is 0, which eventually causes `instruction_o.pc` to be set to 0x0. I previously tried the suggested workaround in #476 and #480, since the issue in #476 looked similar to the issue I was seeing, but the problem still duplicated. I included some relevant waveforms and a snippet the bootrom's object dump.

**Waveforms:**
[image]

**Relevant snippets from objectdump of bootrom:**
[image]
[image]

morenes, 2020-09-17:
hi @JoshAbraham0, I've been reading about your issue and the related issues #476  and #480. Your case look similar, and I think probably the issue is in the previous kill_s2, which causes the state to switch from TAG_CMP to IDLE. Did you apply the change at #480? did u try also the PR from #475?
If those still don't work, if you generate new traces showing also areq_ signals I d be happy to have a look at the waves. From what I've read in the code and the issues abovementioned, the problem is often in the TLB, which makes that the second request gets the translation of the killed request...

morenes, 2020-09-17:
Also probably worth looking at why the target_addr is 0.

MikeOpenHWGroup, 2023-02-17:
Thanks for your insights @morenes.

Hi @JoshAbraham0, thanks for your interest in CVA6.  This issue has not been updated in more than 2.5 years, so I will assume it is resolved and will close this issue.  If you are still having trouble, please feel free to open another.

closed 2023-02-17 by MikeOpenHWGroup