openhwfoundation/cva6 #521: Openpiton+Ariane froze when Execute Region starts from 0

fei-g, 2020-09-07:
Hi Florian,

I don't know how execute region is checked in cva6. We thought that was a check for security, like it will trap when PC is out of the region. We just made the whole memory region executable, and didn't expect this would cause any trouble. But openpiton+ariane froze on FPGA(vc707) during the SD initializing process. We later found that this would happen as long as the execute region starts from 0. Any idea what's the reason?

zarubaf, 2020-09-08:
That is really strange. In the standard configuration, we do have the debug module at the zero page so I am pretty sure that should work. You have an idea if this is really a freeze or is the core continuously trapping? "Freezing" should definitely never happen and would indicate a core bug.

MikeOpenHWGroup, 2023-02-17:
@zarubaf and @fei-g, what is the status of this issue?   It has not been updated for 2.5 years, but it _looks_ serious.

closed 2023-02-17 by MikeOpenHWGroup

Jbalkind, 2023-02-18:
I think this might be partially connected to #927 but I don't recall the exact circumstances. I think it may have indeed been as weird a configuration issue as it sounds.