openhwfoundation/cva6 #525: [m|scounteren] Wrong privilege check

zarubaf, 2020-09-16:
https://github.com/openhwgroup/cva6/blob/0249cfef00de2d6edb8e2dcb3b2fe84cab7a8ef3/src/csr_regfile.sv#L917

I think the condition should be the current privilege level and not the privilege level of the instruction (which will always be U-Mode).

Jbalkind, 2020-09-16:
I wasn't totally clear on what was what when I wrote it so I wouldn't be surprised if I got that wrong 😛

jctullos, 2020-09-25:
Would this affect Linux booting?

Without PMP, the core boots Linux/enters supervisor mode fine. But for some reason, both on 2 builds I've tried (openpiton and ESP), when I use incorporate the PMP changes, Linux hangs right after BBL completes.

Jbalkind, 2020-09-25:
The PMPs haven't been tested in the OP Linux environment at all afaik. I wouldn't be surprised if there was a hangup with them. Also does bbl automatically detect the presence of PMPs or does the code need modified when they're added? OpenSBI detects them automatically at least.

jctullos, 2020-09-25:
Honestly, I'm not 100% on it either. Possibly a privilege issue? It detects them correctly as part of bbl, but it consistently stalls at: (within minit)
  asm volatile ("mret" : : "r" (a0), "r" (a1));

However, I'm not 100% sure if this is where the stall is happening. Nothing else prints out. Is there anything you all would use to debug this with GDB? Spike or QEMU? The tough part is determining if it's a hardware specific, so I might not have any issues with a QEMU VM.

jctullos, 2020-09-26:
Alright, looks like I figured out what was missing. Even though I had CONFIG_HVC_RISCV_SBI=y and earlycon=sbi/console=hvc in the Linux configs, I found one that I had forgotten:

CONFIG_HVC_DRIVER=y
CONFIG_SERIAL_EARLYCON_RISCV_SBI=y

There might have been one other, but I can't remember offhand. But now I have output from Linux again.

Moschn, 2020-09-29:
> Would this affect Linux booting?
> 
> Without PMP, the core boots Linux/enters supervisor mode fine. But for some reason, both on 2 builds I've tried (openpiton and ESP), when I use incorporate the PMP changes, Linux hangs right after BBL completes.

I noticed this last week. Apparently recent changes broke the PMP support again. I am in the process of finding the problem and fixing it. I will also try create a PR with riscv-tests to align their PMP test with any granularity. With this we can re-add the PMP test to the ci of ariane

Moschn, 2023-02-21:
I think this was fixed in #698

closed 2023-02-25 by zarubaf