openhwfoundation/cva6 #534: PMP Linux Issue

jctullos, 2020-09-28:
I've been having issues getting a Linux initramfs build to work using PMP. I believe it's something with the PMP code now, instead of a configuration issue. 

On both OpenPiton and ESP builds, I can get the Linux boot to reach the end before Init runs, and then it just hangs, on both builds. Usually this would be due to an incorrect initramfs build/image. I've tried adding a init that just executes a hello world program and that's not being executed.
The binary runs fine on the same cores without PMP. I've also used a different VM that has successful builds for another PMP Linux distribution, and it also failed. 

Here's where it gets stuck:
[image]

Now with ESP, when I remove the PLIC driver then I can overflow the buffer and it prints another panic for Init that I didn't see before. 

[image]

That might be just related to the buffer overflow, I'm not sure. 

I can't get it to even spawn a terminal shell, which says it's not able to access the folder structure. Any help would be appreciated, thank you!

jctullos, 2020-09-28:
And actually hold off, I might end up closing this comment. I forgot I have a vc707 on hand as well (the other builds were using VCU118). So I'm going to do a vc707 build with PMP and see if it boots before seeing if it's Ariane or not.

Moschn, 2020-09-29:
Which version of Ariane are you using? I noticed that some recent patches have broken the PMP support again.

jctullos, 2020-09-29:
For the VC707 build that I did last night, it was just on the Master. Right now I just checked out: 3663eeb

When I was building the core from the ESP repo for the VCU118, I went all the way back to: efed421

jctullos, 2020-09-29:
@Moschn 

Alright so an update. Using the Ariane SDK:

For the VC707, on branch: eef5ff6 everything boots correctly with PMP into a login. There were some RPC errors earlier when I had the MMC disabled to test initramfs, but it seems like it's resolved now. Either way, vc707 was successful.

I'm rebuilding branch 3663eeb to see if there were any issues. I'll update here. This might just be strictly an issue with VCU118/ESP/Open Piton builds.

Jbalkind, 2020-10-01:
@jctullos if I'd known you had a vc707 I'd have just recommended to use that for OP+Ariane :) it generally works without issue for us

jctullos, 2020-10-02:
@Jbalkind 

You're telling me! I've had the worst trouble with the vcu118. The problem is that on another build, I can't close the timing for the vc707 for another processor. So I'm basically working on both right now and which ever one I can get working first I'll push forward with lol.

jctullos, 2020-10-03:
@Jbalkind 

Great news, got VC707 working on the other processor I was working on that didn't have a build for it. So pushing ahead with the much easier VC707. 

Closing this and glad to never use the VCU118 again. Thanks for all your help on both OpenPiton and Ariane!

closed 2020-10-03 by jctullos

michael-etzkorn, 2021-08-23:
@jctullos I'm not sure if you'll see this, but if you do I'd be interested in getting in contact with you about the OpenPiton build and your vcu118 fork of Cva6. michael-etzkorn@sage.micro.com