openhwfoundation/cva6 #538: PMP risc-v benchmark fails with page fault

Moschn, 2020-10-05:
The official riscv-tests have recently been updated to automatically detect the granularity of the underlying implementation and it would be nice to add them to our CI. However, the PMP benchmark fails with a page fault and AFAIK is unrelated to PMP (Note that this report is based on #537 and #523). If the `INLINE` keyword is removed from the function `test_one` the benchmark/test successfully completes. See patch below:

```patch
diff --git a/benchmarks/pmp/pmp.c b/benchmarks/pmp/pmp.c
index ec07e4a..4a7fbd4 100644
--- a/benchmarks/pmp/pmp.c
+++ b/benchmarks/pmp/pmp.c
@@ -17,8 +17,11 @@ uintptr_t handle_trap(uintptr_t cause, uintptr_t epc, uintptr_t regs[32])
   if (cause == CAUSE_ILLEGAL_INSTRUCTION)
     exit(0); // no PMP support

-  if (!trap_expected || cause != CAUSE_LOAD_ACCESS)
+  if (!trap_expected || cause != CAUSE_LOAD_ACCESS) {
+    printf("trap expected: %d\n", trap_expected);
+    printf("cause: %d, mtval: %p\n", cause, read_csr(mtval));
     exit(1);
+  }
   trap_expected = 0;
   return epc + insn_len(epc);
 }
@@ -97,6 +100,7 @@ INLINE int pmp_ok(pmpcfg_t p, uintptr_t addr, uintptr_t size)
   return hits == 0 || hits >= size;
 }

+// Remove INLINE here and it will work
 INLINE void test_one(uintptr_t addr, uintptr_t size)
 {
   uintptr_t new_mstatus = (read_csr(mstatus) & ~MSTATUS_MPP) | (MSTATUS_MPP & (MSTATUS_MPP >> 1)) | MSTATUS_MPRV;
```

The output from verilator:
```
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 41709
trap expected: 0
cause: 13, mtval: 0x80002dac
*** FAILED *** (tohost = 1)
../orig/riscv-tests/benchmarks/pmp.riscv *** FAILED *** (code = 1) after 370946 cycles
```

I could not figure out why it fails yet (and I am hoping someone with more experience takes over ;) ).

niwis, 2020-10-06:
The responsible code segment looks something like this:
```
set   mstatus.mprv      // enables translation for load/stores
load  virtual_address   // load 1
clear mstatus.mprv      // disables translation for load/stores
load  physical_address  // load 2
```
While the load unit still waits for the D$'s response for `load 1`, it already starts processing `load 2` by translating `physical_address`. Of course, this fails. However, the D$'s response to `load 1` arrives at the same time as the MMU's exception of `load 2`. As a result, the exception of `load 2` retires with `load 1`.
https://github.com/openhwgroup/cva6/pull/540 implements the expected behavior: the translation fault of `load 2` doesn't surface with `load 1`, but `load 2` is successfully retried after translation is disabled.

niwis, 2020-10-06:
Could you confirm whether this resolves the issue?

Moschn, 2020-10-06:
I just tried it and it works like a charm! Thanks!

PR #698 (2021-07-29): pmp: Fix csr
- Add proper granularity (G=1)
- Disallow NA4
- Fix read of pmpaddr
- Bump riscv-tests to support pmp benchmark that detects granularity (https://github.com/riscv/riscv-tests/pull/295)
- Add PMP benchmark to CI

Fixes #538 
Fixes #697
closed 2021-07-29 by zarubaf